YosysHQ/yosys #187: FSM reencoding issues

whitequark, 2016-07-08:
Apologies for the nonspecific issue title, I'm not sure what's the root cause of the bug. I have this code:
```verilog
module top(
        (* LOC="P4"  *) output coilA1,
        (* LOC="P6"  *) output coilA2,
        (* LOC="P8"  *) output coilB1,
        (* LOC="P10" *) output coilB2
    );

    wire strobe;
    GP_LFOSC #(
        .OUT_DIV(8'd16),
        .AUTO_PWRDN(1'b0)
    ) lfosc (
        .CLKOUT(strobe)
    );

    localparam STATE0 = 4'b0100,
               STATE1 = 4'b0001,
               STATE2 = 4'b1000,
               STATE3 = 4'b0010;
    // (* fsm_encoding="auto" *)
    reg [3:0] state;
    always @(posedge strobe)
        case(state)
            STATE0:  state <= STATE1;
            STATE1:  state <= STATE2;
            STATE2:  state <= STATE3;
            STATE3:  state <= STATE0;
            default: state <= STATE0;
        endcase

    assign {coilA1, coilA2, coilB1, coilB2} = state;

endmodule
```

It works, but translates into lots of LUTs. If I uncomment the fsm_encoding attribute, it is translated into four DFFs, which is what I want, but it doesn't work (all outputs are low) anymore because all of them are initialized to 0. If I explicitly initialize it to `STATE0` (and comment out the `default` case), then it is translated into eight DFFs, and, most interestingly, still doesn't work (all outputs stay low), despite one of the DFFs being initialized to 1.

I'm synthesizing using `yosys -p "synth_greenpak4 -part SLG46620V -json test.txt" test.v`.

I'm not sure what should happen according to the standard in the case where `reg` is uninitialized (though @azonenberg thinks it ought be the same as without the FSM annotation), but the second case seems clearly wrong.

cliffordwolf, 2016-07-08:
The important insight here is that the `fsm_detect` pass does not set the `fsm_encoding` on the state signal automatically. This is probably because the pass thinks it's not save to apply the FSM optimization passes to this state machine. Your experiments seem to confirm this, showing that the decision `fsm_detect` made initially was correct.

> It works, but translates into lots of LUTs.

Have you considered something like the following?

    reg [3:0] state;

    always @(posedge strobe)
          state <= {state[2:0], !state[2:0]};

    assign {coilA1, coilA2, coilB1, coilB2} = {state[2], state[0], state[3], state[1]};

This translates to 1x GP_3LUT, 2x GP_DFF, and 1x GP_SHREG (or two more GP_DFF instead of the GP_SHREG, if there are better things to do for the GP_SHREG cells).

I'm closing this issue now. If you still feel like there is a bug, just say so and I will reopen it.

(You'd either need to use `reg [3:0] state = 0;` or `state[2:0] === 0` instead of `!state[2:0]` for it to start properly in simulation. This variations all synthesize to the same circuit. I think initializing `state = 0` is the cleanest way to fix simulation in this case.)

whitequark, 2016-07-08:
Yes, I do think there is still a bug--at the very least a documentation bug. I did not know that using the fsm_encoding attribute can lead to yosys silently miscompiling code, and there is nothing in the manual to suggest that it can be; indeed, the manual explicitly suggests overriding negative decisions of `fsm_detect`:

> It is possible to overwrite it by setting \fsm_encoding= "auto"
on registers that should be considered FSM state registers and setting \fsm_encoding= "none" on registers
> that match the above criteria but should not be considered FSM state registers.

Otherwise--I have found that writing down the FSM in the most natural way and then manipulating the encoding through Yosys is rather convenient (in cases less trivial than the above). Is there a fundamental reason why the 2nd example (with `(*fsm_encoding="auto"*)` and explicit register initialization) cannot be correctly translated?

cliffordwolf, 2016-07-08:
I've added an additional remark about that to the manual in commit d3f0d72.

> Is there a fundamental reason why the 2nd example (with `(*fsm_encoding="auto"*)` and explicit register initialization) cannot be correctly translated?

In order for the FSM passes to work there must be a detectable reset signal, and the FSM should start uninitialized. The reset signal does not need to be a synchronous reset. An init value on the reset register is ignored. That's why this is also a criteria for not considering a register for FSM state optimization:

https://github.com/cliffordwolf/yosys/blob/d3f0d7242732d8fb8885600c00c80c1aff8b5253/passes/fsm/fsm_detect.cc#L113

So this is why explicit initialization doesn't fix it: The FSM framework simply ignores such an init value.

Your FSM does not have a reset signal that forces the FSM into a known state, instead you rely on the FSM detecting an invalid state and somehow using that to go into a valid state instead. This is not representable using `$fsm` cells. The framework only works if it can assume that the FSM is never in an invalid state, except before the reset signal is asserted, in which case it has undefined behavior before that.

Or put in another way: When you rely on your FSM always being in a valid state, or somehow need to distinguish between valid and invalid states, then you introduce restrictions on what the state encoding may look like. But the whole point of FSM optimization is that the synthesis tool can choose an encoding at will without restrictions.

(I'm writing this while a hyperactive child is yelling into my right ear... I hope what I'm writing makes at least some sense. If something is unclear, keep asking and I will get back to it when I'm undisturbed.)

whitequark, 2016-07-08:
This makes a lot of sense, thanks! I assume I will get the result I expect if I add an explicit reset. Unfortunately, this slightly pessimizes GreenPAK as not all DFFs have a set/reset input, but this is probably not especially important.

closed 2016-07-08 by whitequark