YosysHQ/yosys #188: Feature request: FSM init fixes

azonenberg, 2016-07-09:
I understand the need for the fsm pass to understand the default state. But for some architectures (Xilinx FPGAs for example) external resets add extra logic and reduce packing density compared to only specifying an initial state (assuming no runtime reset is necessary). For others (Silego GreenPak) only some FFs have reset capability, while all can be initialized at POR.

My suggestion is to allow FSM registers to be initialized according to the following rules:
- Reset, no init value: existing behavior
- Reset, plus initialized at POR: existing behavior plus propagate INIT attribute (updated to re-coded default state). Possibly disallow having init and reset with different values.
- No reset, POR value specified: propagate re-coded INIT attribute, use this as default state

In addition, I think that silently generating incorrect code is always a bug, period. If you specify fsm encoding in your RTL and Yosys cannot match the register to a supported FSM inference structure, this should be an error or at least a warning. Look at what I did for the greenpak4_counters pass for an example (forcing counter extraction when the logic cannot map to hard IP fails).

cliffordwolf, 2016-07-09:
For reference: #187

cliffordwolf, 2016-07-09:
Commits d63ffab and b5a9fba improve the verbosity of `fsm_detect`, improve detection of self-resetting FSMs, and add warning messages for state wires with `fsm_encoding` attributes on them when that might cause unexpected problems.

Given the state of open todo items and my current list of priorities this is pretty much as far as I'm willing to go right now. Feel free to create a PR that improves handling of things like self-resetting FSMs.

(Closing the issue now. I'll reopen it on request, but don't expect me to spend much time on it.)

> [..] this should be an error [..]

This would require the `fsm_detect` pass to be able to make a definitive decision on whether running the FSM optimizations will actually cause a problem. Feel free to formulate an algorithm that does this for all corner cases and implement it. Right now all I can offer is a heuristic, and it is possible that the heuristic is too pessimistic and that the user actually knows it better. Therefore it can only be a warning at the moment.

whitequark, 2016-07-11:
@cliffordwolf First off, I appreciate these commits quite a bit (especially the fsm_detect verbosity), and I think they go in the right direction. But, unfortunately, they can introduce silent miscompilation. E.g. this code:

```verilog
    reg [1:0] cycle;
    always @(posedge i_clk)
        case(cycle)
            2'd0: cycle <= 2'd1;
            2'd1: cycle <= 2'd2;
            2'd2: cycle <= 2'd0;
        endcase
```
is, as far as I can see, has perfectly defined behavior for synthesis, with implicit initialization of `cycle` to `2'b00`. In current master however, yosys will autodetect this as an FSM and convert it to one-hot, but not update the init values to reflect the reset state, so the one-hot FSM becomes completely cold at POR. This seems like a clear miscompilation to me.

The silliest part is that yosys actually *correctly detects the reset state*:
```
  State encoding:
    0:      3'--1  <RESET STATE>
    1:      3'-1-
    2:      3'1--
```
but it doesn't bother to reflect that in the reencoded one-hot FSM.

While we're at it, do you think you could infer the reset state from the initialization value if there's no explicit reset (and just refuse to reencode FSMs that use both)? It sounds trivial to do; one condition that uses the data that you already have elsewhere.

While I was looking at this topic I discovered that Xilinx also says in [WP272](http://www.xilinx.com/support/documentation/white_papers/wp272.pdf) that it is almost always a bad idea to use explicit reset for initialization in FPGAs, for both resource usage and timing.

If you are really busy I guess I could implement this myself, though I'm unfamiliar with the topic and so I'm wary of introducing other, more obscure miscompilations somewhere.

cliffordwolf, 2016-07-12:
>  [..] as far as I can see, has perfectly defined behavior for synthesis, with implicit initialization of cycle to 2'b00 [..]

There is no implicit initialization of anything to anything in synthesis. That's not how synthesis works. This code has undefined behavior in simulation (cycle will be stuck at 2'bxx). When it has undefined behavior in simulation, the synthesis tool can do whatever it wants.

The original circuit does not self-reset in this case. So it is "safe" to create a non-self-resetting FSM from it.

> The silliest part is that yosys actually correctly detects the reset state:

No, it doesn't. It just picks a reset state at random.

> While we're at it, do you think you could infer the reset state from the initialization value if there's no explicit reset (and just refuse to reencode FSMs that use both)?

This is the easiest part. But I also have to remember that the reset value comes from an initial value, so I have to change the FSM representation (the $fsm cell type) and update all users of that cell type to understand the new format and handle it correctly, and of course I also have to re-create the circuit with that kind of reset value correctly. And I have to update the FSM test bench (tests/fsm/) to create and check such circuits.

I'll leave this ticket open and see when I will get to it. Don't hold your breath.

whitequark, 2016-07-12:
> This code has undefined behavior in simulation (cycle will be stuck at 2'bxx). When it has undefined behavior in simulation, the synthesis tool can do whatever it wants.

OK, I see, I've missed that part. I understand that you normally recommend to simulate everything first (which would have caught this) but Silego is somewhat of an odd case because it currently lacks behavioral definitions for almost all cells and I'm not even sure if iverilog can do all the mixed-signal stuff Silego uses. The vendor's tools don't have any simulation capability either.

So, this wasn't a miscompilation, but I think there's still opportunity for improvement. I'll try to think of something reasonably unobtrustive.