YosysHQ/yosys #190: Autotest.sh and part select

Kmanfi, 2016-07-11:
Run with:
`autotest.sh -e -p 'prep -ifx' part_sel.v`

```
module part_sel
(
   lego0,
   lego0foo_i,
   clk,
   rst
);

   output [0:63] lego0;	
   input [0:63] lego0foo_i;

   input clk;
   input rst;

   reg [0:63] lego0_q;

   integer i0, i1, i2, i3;

   always @(posedge clk) begin
    if (rst) begin
	 for (i0=0; i0<=15; i0=i0+1) begin
		lego0_q[i0] <= 0;
	 end
    end
    else begin
	 for (i0=0; i0<=15; i0=i0+1) begin
		     lego0_q[i0] <= lego0_q[i0];
	 end
    end
   end

   always @(posedge clk) begin
    if (rst) begin
	 for (i1=16; i1<=31; i1=i1+1) begin
		lego0_q[i1] <= 0;
	 end
    end
    else begin
	 for (i1=16; i1<=31; i1=i1+1) begin
		     lego0_q[i1] <= lego0_q[i1];
	 end
	end
   end

   always @(posedge clk) begin
    if (rst) begin
	 for (i2=32; i2<=47; i2=i2+1) begin
		lego0_q[i2] <= 0;
	 end
    end
    else begin
	 for (i2=32; i2<=47; i2=i2+1) begin
		     lego0_q[i2] <= lego0_q[i2];
	 end
	end
   end

   always @(posedge clk) begin
    if (rst) begin
	 for (i3=48; i3<=63; i3=i3+1) begin
		lego0_q[i3] <= 0;
	 end
    end
    else begin
	 for (i3=48; i3<=63; i3=i3+1) begin
		lego0_q[i3] <= lego0foo_i[i3];
	 end
    end
   end
      
   assign lego0  = lego0_q;
   
endmodule 
```

Kmanfi, 2016-07-25:
It seems that module inputs and outputs are 63:0 in _syn0 instead of 0:63.

Kmanfi, 2016-08-02:
Autotest.sh give ok without -e parameter.
```
autotest.sh -p 'prep -ifx' part_sel.v
Test: part_sel -> ok
autotest.sh -e -p 'prep -ifx' part_sel.v
Test: part_sel -> ERROR!
```

Kmanfi, 2016-08-15:
Tested also with vsim. Same behaviour as in my previous comment.
It seems Yosys generates non valid Verilog (_syn0) with the above code and -e option.

cliffordwolf, 2016-08-15:
Fixed in commit f0a8713.

closed 2016-08-15 by cliffordwolf