YosysHQ/yosys #199: Handling GP_DFF*I cells for synthesis

whitequark, 2016-08-14:
GreenPAK 4 GP_DFF*I cells are the same as GP_DFF* cells, but they only have an inverting nQ output. As far as I can tell, dfflibmap can only map $_DFF_N_ to a single type of cell, inverting or not.

Ideally, what would happen is that for a FF cell, if the Q output has higher fanout than nQ, then it is synthesized as GP_DFF*, otherwise as GP_DFF*I. But as a first approximation, converting GP_DFF* cells that only drive a NOT to GP_DFF*I also works.

azonenberg, 2016-08-14:
Longer term the inversion should work at both levels of the chain - an inverter that only drives a GP_DFF* should be merged into a GP_DFF*I.

Note that this will require mapping GP_DFFS to GP_DFFRI and vice versa, since we're re-timing the inverter from before the FF to after it.

cliffordwolf, 2016-08-15:
The planned `$_FNOT_` feature that I mentioned during our original conversation on twitter actually would not help much in this case: It will be more useful  for DFF cell types that have both inverted and non-inverted outputs.

I've now added a `greenpak4_dffinv` pass in commit 5d90a5b that should do the right thing in most cases. (In the remaining cases you'd get a correct circuit but with an extra inverter. This could be fixed by adding a feature that inverts LUT inputs by rewriting the LUT when a FF output signal is fed into a `GP_INV` as well as one or more `GP_?LUT` cells).

I have not tested this new pass very well. I'll leave it to you to test it. If you find an example that breaks it, just post the verilog code here and I'll see if I can fix it.

Re @azonenberg's comment: I think mapping from e.g. GP_DFFS to GP_DFFRI must be performed when merging an inverter on the FF _output_ side, when merging an input inverter this should not effect the SET/RESET path or the INIT value of the FF. At least that's how I implemented it in the new pass.

closed 2016-08-15 by cliffordwolf

whitequark, 2016-08-20:
> I have not tested this new pass very well. I'll leave it to you to test it.

FYI I did some cursory checks and it performed well so far.