YosysHQ/yosys #204: Make Error

justinrf, 2016-08-17:
Using GCC to make Yosys I get the following error:

[100%] Building yosys
[100%] Building yosys-config
[100%] Building abc/abc-ae7d65e71adc
Pulling ABC from bitbucket.org:
+ test -d abc
+ hg clone https://bitbucket.org/alanmi/abc abc
/bin/sh: line 3: hg: command not found
+ cd abc
/bin/sh: line 4: cd: abc: No such file or directory
make: *** [abc/abc-ae7d65e71adc] Error 1

Any ideas how to fix this?

Kind regards,

Justin Fisher.

udif, 2016-08-17:
You failed because you do not have mercurial (hg) installed . The copy of ABC Yosys is using is tored on a Mercurial version control system under bitbucket.org

I don't know what system you use, but on a Ubuntu you probably need to do 'apt-get install mercurial'

cliffordwolf, 2016-08-17:
What @udif said. Quoting the README file:

```
For example on Ubuntu Linux 14.04 LTS the following commands will install all
prerequisites for building yosys:

	$ yosys_deps="build-essential clang bison flex libreadline-dev gawk
	       tcl-dev libffi-dev git mercurial graphviz xdot pkg-config python3"
	$ sudo apt-get install $yosys_deps
```

closed 2016-08-17 by cliffordwolf