YosysHQ/yosys #207: autotest.sh and mem23

Kmanfi, 2016-08-19:
Hi, I sent code to your private email.

cliffordwolf, 2016-08-19:
The problem here is that you are using a 8 bit address to write to a memory that is defined as [0:31]. In simulation out-of-bounds memory writes are simply ignored, in synthesis however only the lower 5 bits of the address are used, so e.g. a write to address 32 ends up as write to address 0. This is a usual approach to memories in synthesis (and a common cause for simulation-synthesis-mismatches in many projects).

Declare the memory as [0:255] instead of [0:31], or mask the address on memory writes, e.g. using `[JmpTabWrAdrs & 31]` instead of `[JmpTabWrAdrs]`, to solve the problem.

closed 2016-08-19 by cliffordwolf

cliffordwolf, 2016-08-19:
I've now added support for `prep -memx`, which emulates the simulation behavior for out-of-bounds memory accesses. This should help identifying such problems in verilog code. (commit 15ef608)