YosysHQ/yosys #210: Bug in Verilog frontend

adamwalker, 2016-08-21:
I believe that Yosys (git SHA a93fcec) fails to handle the following example correctly:
```
module Main_fir_sfir(result);
  output signed [9:0] result;
  
  wire signed [9:0] intermediate [0:2];
  assign intermediate[0] = 1;
  assign intermediate[1] = 2;
  
  function integer depth2Index;
    input integer depth;
  
    depth2Index = depth;
  endfunction
  
  assign intermediate[depth2Index(2)] = intermediate[0] + intermediate[1];  
  assign result = intermediate[2];
endmodule
```
Specifically, when I show the schematic using the show command, intermediate[2] is not driven, but it should be driven by the second to last line of the module. As a result, when I optimise it with opt_clean -purge, the whole circuit disappears.

This is a simplification of the Verilog generated by the [Clash](http://www.clash-lang.org/) compiler for the FIR filter example on that page.

It works correctly if I replace the depthToIndex(2) call with the literal 2. i.e. I remove the depthToIndex function call on the second to last line.

cliffordwolf, 2016-08-21:
Thanks for the bug report and test case. That problem is fixed now in commit 82a4a02.

The mechanism in Yosys that handles Verilog arrays that are not describing memories (mem2reg) is complicated and there are many strange corner cases. At the same time it's not very well tested because most (hand written) Verilog code avoids constructs like that. I would not be surprised if you'd find more bugs there..

closed 2016-08-21 by cliffordwolf

adamwalker, 2016-08-21:
Thanks for the quick fix!
Looking at the schematic, it appears to be synthesizing the full FIR example correctly now.