YosysHQ/yosys #214: RAM4K earlier address test

nononame, 2016-08-26:
I have a simple description of a ROM with 12 addresses (not a power of 2).

A first component describing an asynchronous ROM ([myrom_success.v](https://www.le-huit.fr/ephemeral/yosysrom/myrom_success.html) ([raw](https://www.le-huit.fr/ephemeral/yosysrom/myrom_success.v)) or [myrom_failure.v](https://www.le-huit.fr/ephemeral/yosysrom/myrom_failure.html) ([raw](https://www.le-huit.fr/ephemeral/yosysrom/myrom_failure.v))), a read register attached to this asynchronous ROM in the top component: [top.v](https://www.le-huit.fr/ephemeral/yosysrom/top.html) ([raw](https://www.le-huit.fr/ephemeral/yosysrom/top.v).

Yosys successfully run with [myrom_success.v](https://www.le-huit.fr/ephemeral/yosysrom/myrom_success.html) ([raw](https://www.le-huit.fr/ephemeral/yosysrom/myrom_success.v)) and detect the RAM4K:

~~~
yosys -Q  -p \
  "synth_ice40 -abc2 -top top -blif success.blif; write_verilog success.v" \
  myrom_success.v top.v
~~~

Due to some simulation metastability problems, I have to prevent the case _address >= 12_: [myrom_failure.v](https://www.le-huit.fr/ephemeral/yosysrom/myrom_failure.html) ([raw](https://www.le-huit.fr/ephemeral/yosysrom/myrom_failure.v)).

~~~
yosys -Q  -p \
  "synth_ice40 -abc2 -top top -blif failure.blif; write_verilog failure.v" \
  myrom_failure.v top.v
~~~

This is done by adding a test for the data output of the asynchronous ROM. **Yosys fails** with this added test:

~~~
3.8. Executing MEMORY_BRAM pass (mapping $mem cells to block memories).
Processing top.myrom_i.rom:
  Properties: ports=1 bits=132 rports=1 wports=0 dbits=12 abits=4 words=11
  Checking rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1):
    Bram geometry: abits=8 dbits=16 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M0: awaste=245 dwaste=4 bwaste=3964 waste=3964 efficiency=3
    Rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1) accepted.
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Read port #0 is in clock domain \clk.
        Mapped to bram port A1.1.
      Updated properties: dups=1 waste=3964 efficiency=3
      Storing for later selection.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 1):
    Bram geometry: abits=9 dbits=8 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=501 dwaste=4 bwaste=4052 waste=4052 efficiency=1
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 1) rejected: requirement 'min efficiency 2' not met.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 2):
    Bram geometry: abits=10 dbits=4 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=1013 dwaste=0 bwaste=4052 waste=4052 efficiency=1
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 2) rejected: requirement 'min efficiency 2' not met.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 3):
    Bram geometry: abits=11 dbits=2 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=2037 dwaste=0 bwaste=4074 waste=4074 efficiency=0
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 3) rejected: requirement 'min efficiency 2' not met.
  Selecting best of 1 rules:
    Efficiency for rule 1.1: efficiency=3, cells=1, acells=1
    Selected rule 1.1 with efficiency 3.
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Read port #0 is in clock domain \clk.
        Mapped to bram port A1.1.
      Creating $__ICE40_RAM4K_M0 cell at grid position <0 0 0>: myrom_i.rom.0.0.0
        Adding extra logic for transparent port A1.1.
ERROR: Assert `it.first.size() == it.second.size()' failed in kernel/rtlil.cc:1191.
~~~

- Yosys version: `Yosys 0.6+234 (git sha1 4be4969, clang 3.5.0-10 -fPIC -Os)`
- All files in an archive: [yosysrom.tar.gz](https://www.le-huit.fr/ephemeral/yosysrom.tar.gz)

cliffordwolf, 2016-08-27:
Thanks for the bug report. Fixed in commit 23afead.

closed 2016-08-27 by cliffordwolf

nononame, 2016-08-28:
I confirm that it is fixed. Thanks!