YosysHQ/yosys #217: Building on Fedora 24

bkmgit, 2016-08-28:
Current git version and release 0.6 have trouble building on Fedora 24, though release 0.5 seems to build without trouble (http://koji.fedoraproject.org/koji/taskinfo?taskID=15415187). Would be helpful to also get this into Fedora repositories. Let me know if have any suggestions for the error I get below when trying to make an rpm build version 0.6 is:

[ 40%] Building passes/cmds/qwp.o
passes/cmds/qwp.cc:287:21: error: use of undeclared identifier 'fabs'
                                double absval = fabs(M[i + row*N1]);
                                                ^
passes/cmds/qwp.cc:344:13: error: no member named 'isfinite' in namespace 'std';
      did you mean 'isdigit'?
                        if (std::isfinite(v)) {
                            ~~~~~^~~~~~~~
                                 isdigit
/usr/bin/../lib/gcc/x86_64-redhat-linux/6.1.1/../../../../include/c++/6.1.1/cctype:67:11: note: 
      'isdigit' declared here
  using ::isdigit;
          ^
passes/cmds/qwp.cc:526:14: error: no member named 'isfinite' in namespace 'std';
      did you mean 'isdigit'?
                                if (std::isfinite(node.pos)) {
                                    ~~~~~^~~~~~~~
                                         isdigit
/usr/bin/../lib/gcc/x86_64-redhat-linux/6.1.1/../../../../include/c++/6.1.1/cctype:67:11: note: 
      'isdigit' declared here
  using ::isdigit;
          ^
passes/cmds/qwp.cc:533:14: error: no member named 'isfinite' in namespace 'std';
      did you mean 'isdigit'?
                                if (std::isfinite(node.pos)) {
                                    ~~~~~^~~~~~~~
                                         isdigit
/usr/bin/../lib/gcc/x86_64-redhat-linux/6.1.1/../../../../include/c++/6.1.1/cctype:67:11: note: 
      'isdigit' declared here
  using ::isdigit;
          ^
passes/cmds/qwp.cc:673:7: error: use of undeclared identifier 'fabs'
                if (fabs(max_value - min_value) < 0.001) {
                    ^
passes/cmds/qwp.cc:730:27: error: use of undeclared identifier 'pow'
                        double distance = sqrt(pow(node1.pos - node2.pos...
                                               ^
passes/cmds/qwp.cc:730:59: error: use of undeclared identifier 'pow'
  ...double distance = sqrt(pow(node1.pos - node2.pos, 2) + pow(node1.alt_po...
                                                            ^
7 errors generated.
Makefile:339: recipe for target 'passes/cmds/qwp.o' failed
make: *** [passes/cmds/qwp.o] Error 1
error: Bad exit status from /var/tmp/rpm-tmp.B2jIFX (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.B2jIFX (%build)

cliffordwolf, 2016-08-28:
Installing Fedora 24 in a VM now. Some questions:

1. What packages do I need to `dnf install` to get to a similar setup to what you have?

2. Are you using a GCC or CLANG build? I.e. which build config (Makefile.conf, etc) are you using to build Yosys?

bkmgit, 2016-08-28:
Example spec file is below. At the moment uses CLANG - will try GCC.  Can later add rpm packages in copr (https://copr.fedorainfracloud.org/)

Name: yosys
Version: 66582964bc11aadf3d0783a346706d801451a13f  
Release: 1%{?dist}
Summary: Open Synthesis Suite

License: Yosys is free software licensed under the ISC license (a GPL compatible license that is similar in terms to the MIT license or the 2-clause BSD license).
URL: http://www.clifford.at/yosys/
Source0: https://github.com/cliffordwolf/yosys/archive/master.zip 
ExclusiveArch: x86_64

BuildRequires: abc-devel
BuildRequires: abc
BuildRequires: abc-libs
BuildRequires: readline-devel
BuildRequires: bison
BuildRequires: bison-devel
BuildRequires: flex
BuildRequires: flex-devel
BuildRequires: libffi-devel
BuildRequires: mercurial
BuildRequires: git
BuildRequires: clang
BuildRequires: gcc
BuildRequires: python3
BuildRequires: tcl-devel
BuildRequires: graphviz-devel
BuildRequires: graphviz-tcl
BuildRequires: gawk
BuildRequires: python-xdot
BuildRequires: iverilog

Requires: abc
Requires: abc-libs
Requires: iverilog
%description
This is a framework for RTL synthesis tools. It currently has
extensive Verilog-2005 support and provides a basic set of
synthesis algorithms for various application domains.

Yosys can be adapted to perform any synthesis job by combining
the existing passes (algorithms) using synthesis scripts and
adding additional passes as needed by extending the yosys C++
code base.

Yosys is free software licensed under the ISC license (a GPL
compatible license that is similar in terms to the MIT license
or the 2-clause BSD license).

%prep
rm -rf yosys-master
unzip %{SOURCE0}
cd yosys-master
mkdir abc
mkdir abc/bin
mkdir abc/lib64
ln -s /usr/lib64/libabc.so  $PWD/abc/lib64/libabc.so
ln -s /usr/lib64/libabc.so.0 $PWD/abc/lib64/libabc.so.0
ln -s /usr/lib64/libabc.so.0.0.0 $PWD/abc/lib64/libabc.so.0.0.0
ln -s /usr/bin/abc $PWD/abc/bin

sed -i.bak 's_PREFIX ?= /usr/local_PREFIX ?= /usr/_' Makefile
sed -i.bak 's_ABCREV = ae7d65e71adc_ABCREV = default_' Makefile
sed -i.bak 's_ABCEXTERNAL =_ABCEXTERNAL = /usr/bin/abc_' Makefile
sed -i.bak 's_ABCPULL = 1_ABCPULL = 0_' Makefile
sed -i.bak 's/ENABLE_ABC := 1/ENABLE_ABC := 0/' Makefile

cd ..

%build
cd yosys-master
make
cd ..

%install
cd yosys-master
make install DESTDIR=%{buildroot}
cd ..

%check

%clean
rm -rf %{buildroot} 

%files
/bin/yosys
/bin/yosys-config
/bin/yosys-filterlib
/share/yosys/adff2dff.v
/share/yosys/blackbox.v
/share/yosys/cells.lib
/share/yosys/include/backends/ilang/ilang_backend.h
/share/yosys/include/kernel/celltypes.h
/share/yosys/include/kernel/consteval.h
/share/yosys/include/kernel/hashlib.h
/share/yosys/include/kernel/log.h
/share/yosys/include/kernel/macc.h
/share/yosys/include/kernel/modtools.h
/share/yosys/include/kernel/register.h
/share/yosys/include/kernel/rtlil.h
/share/yosys/include/kernel/satgen.h
/share/yosys/include/kernel/sigtools.h
/share/yosys/include/kernel/utils.h
/share/yosys/include/kernel/yosys.h
/share/yosys/include/libs/ezsat/ezminisat.h
/share/yosys/include/libs/ezsat/ezsat.h
/share/yosys/include/libs/sha1/sha1.h
/share/yosys/include/passes/fsm/fsmdata.h
/share/yosys/pmux2mux.v
/share/yosys/simcells.v
/share/yosys/simlib.v
/share/yosys/techmap.v
/share/yosys/xilinx/arith_map.v
/share/yosys/xilinx/brams.txt
/share/yosys/xilinx/brams_map.v
/share/yosys/xilinx/cells_map.v
/share/yosys/xilinx/cells_sim.v

%changelog
* Sun Aug 28 2016 Benson Muite <benson_muite@emailplus.org>
- Initial version of package, make test fails at present, needs iverilog from git

cliffordwolf, 2016-08-28:
I've installed Fedora 24 in a VM and I can build `passes/cmds/qwp.o` with GCC and CLANG without getting the error you quoted. Unfortunately I've created a too small virtual disk and I run out of disk space when trying to build all of Yosys. But I can do enough to tell you that I can't reproduce the issue.

Instructions to reproduce what I am seeing:

I've installed Fedora 24 using this iso image: http://mirror.karneval.cz/pub/linux/fedora/linux/releases/24/Workstation/x86_64/iso/Fedora-Workstation-Live-x86_64-24-1.2.iso

I have installed all of the build requirements in the spec file you have posted, with the exception of the `abc*` packages. In addition I installed `gcc-c++` and `@development-tools`:

```
sudo dnf install @development-tools
sudo dnf install readline-devel bison bison-devel flex flex-devel \
                 libffi-devel mercurial git clang gcc gcc-c++ \
                 python3 tcl-devel graphviz-devel graphviz-tcl \
                 gawk python-xdot iverilog 
```

Then I cloned current git head and built `qwp.o`:

```
git clone http://github.com/cliffordwolf/yosys
cd yosys

rm -f passes/cmds/qwp.o
make PRETTY=0 CONFIG=gcc passes/cmds/qwp.o

rm -f passes/cmds/qwp.o
make PRETTY=0 CONFIG=clang passes/cmds/qwp.o
```

Here is a screenshot:

[image]

I'd say there is something messed up with your installation, maybe with the libstdc++ headers your clang is using.

You can try making a fresh Fedora 24 install in a VM and see if it works there. If it does, see whats different in the VM and your host system.

I'm closing this issue for now. Feel free to post any new information and I'll reopen the issue if there's something new I should look into.

closed 2016-08-28 by cliffordwolf

bkmgit, 2016-08-28:
Sorry, was not clear. Error occurs with release 0.6 - updated build options as suggested, build is done in virtual machine with clean environment. With current version head, build seems ok, but installation step fails with:

Compiling /home/benson/rpmbuild/BUILDROOT/yosys-66582964bc11aadf3d0783a346706d801451a13f-1.fc24.x86_64/usr/share/yosys/python3/smtio.py ...
  File "/usr/share/yosys/python3/smtio.py", line 97
    print(stmt, file=self.debug_file)
                    ^
SyntaxError: invalid syntax

error: Bad exit status from /var/tmp/rpm-tmp.UJ9Bpr (%install)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.UJ9Bpr (%install)

cliffordwolf, 2016-08-28:
```
Compiling /home/benson/rpmbuild/BUILDROOT/yosys-66582964bc11aadf3d0783a346706d801451a13f-1.fc24.x86_64/usr/share/yosys/python3/smtio.py ...
  File "/usr/share/yosys/python3/smtio.py", line 97
    print(stmt, file=self.debug_file)
                    ^
SyntaxError: invalid syntax
```

I can only assume this comes from some generic RPM builder code. (It's not from the yosys Makefile.) Looks like it's trying to compile my python3 scripts with python2.

bkmgit, 2016-08-29:
Thanks, it built after making it use python 3. A repository for x86_64 is here:
https://copr.fedorainfracloud.org/coprs/fed500/Yosys/
Spec file below, still needs more cleanup. Would it be worth adding to repository, then copr can do automatic builds for Fedora and RHEL? Can fork and then make a pull request, but if have a specific place you would like to have the spec file let me know or make it beforehand.

Name: yosys
Version: 66582964bc11aadf3d0783a346706d801451a13f  
Release: 1%{?dist}
Summary: Open Synthesis Suite

License: Yosys is free software licensed under the ISC license (a GPL compatible license that is similar in terms to the MIT license or the 2-clause BSD license).
URL: http://www.clifford.at/yosys/
Source0: https://github.com/cliffordwolf/yosys/archive/master.zip 
ExclusiveArch: x86_64
%global __python %{__python3}

BuildRequires: gettext
BuildRequires: cvs
BuildRequires: diffstat
BuildRequires: doxygen
BuildRequires: patch
BuildRequires: patchutils
BuildRequires: rcs
BuildRequires: systemtap
BuildRequires: gcc-c++
BuildRequires: readline-devel
BuildRequires: bison
BuildRequires: bison-devel
BuildRequires: flex
BuildRequires: flex-devel
BuildRequires: libffi-devel
BuildRequires: gcc
BuildRequires: python3
BuildRequires: python3-devel
BuildRequires: tcl-devel
BuildRequires: graphviz-devel
BuildRequires: graphviz-tcl
BuildRequires: gawk
BuildRequires: iverilog

Requires: iverilog
Requires: python3

%description
This is a framework for RTL synthesis tools. It currently has
extensive Verilog-2005 support and provides a basic set of
synthesis algorithms for various application domains.

Yosys can be adapted to perform any synthesis job by combining
the existing passes (algorithms) using synthesis scripts and
adding additional passes as needed by extending the yosys C++
code base.

Yosys is free software licensed under the ISC license (a GPL
compatible license that is similar in terms to the MIT license
or the 2-clause BSD license).

%prep
rm -rf yosys-master
unzip %{SOURCE0}
cd yosys-master
make config-gcc

sed -i.bak 's_PREFIX ?= /usr/local_PREFIX ?= /usr/_' Makefile
sed -i.bak 's_ABCREV = ae7d65e71adc_ABCREV = default_' Makefile
sed -i.bak 's_ABCEXTERNAL =_ABCEXTERNAL = /usr/bin/abc_' Makefile
sed -i.bak 's_ABCPULL = 1_ABCPULL = 0_' Makefile
cd ..

%build
cd yosys-master
make CONFIG=gcc
cd ..

%check

%install
cd yosys-master
mkdir -p %{buildroot}/bin
install yosys %{buildroot}/bin
install yosys-config %{buildroot}/bin
install yosys-filterlib %{buildroot}/bin
install yosys-smtbmc %{buildroot}/bin
mkdir -p %{buildroot}/share/yosys
cp -r share/. %{buildroot}/share/yosys
cd ..

%clean
rm -rf %{buildroot} 

%files
/bin/yosys
/bin/yosys-config
/bin/yosys-filterlib
/bin/yosys-smtbmc
/share/yosys/adff2dff.v
/share/yosys/cells.lib
/share/yosys/include/backends/ilang/ilang_backend.h
/share/yosys/include/kernel/celltypes.h
/share/yosys/include/kernel/consteval.h
/share/yosys/include/kernel/hashlib.h
/share/yosys/include/kernel/log.h
/share/yosys/include/kernel/macc.h
/share/yosys/include/kernel/modtools.h
/share/yosys/include/kernel/register.h
/share/yosys/include/kernel/rtlil.h
/share/yosys/include/kernel/satgen.h
/share/yosys/include/kernel/sigtools.h
/share/yosys/include/kernel/utils.h
/share/yosys/include/kernel/yosys.h
/share/yosys/include/libs/ezsat/ezminisat.h
/share/yosys/include/libs/ezsat/ezsat.h
/share/yosys/include/libs/sha1/sha1.h
/share/yosys/include/passes/fsm/fsmdata.h
/share/yosys/pmux2mux.v
/share/yosys/simcells.v
/share/yosys/simlib.v
/share/yosys/techmap.v
/share/yosys/xilinx/arith_map.v
/share/yosys/xilinx/brams.txt
/share/yosys/xilinx/brams_map.v
/share/yosys/xilinx/cells_map.v
/share/yosys/xilinx/cells_sim.v
/share/yosys/greenpak4/cells_map.v
/share/yosys/greenpak4/cells_sim.v
/share/yosys/greenpak4/gp_dff.lib
/share/yosys/ice40/arith_map.v
/share/yosys/ice40/brams.txt
/share/yosys/ice40/brams_init1.vh
/share/yosys/ice40/brams_init2.vh
/share/yosys/ice40/brams_init3.vh
/share/yosys/ice40/brams_map.v
/share/yosys/ice40/cells_map.v
/share/yosys/ice40/cells_sim.v
/share/yosys/ice40/latches_map.v
/share/yosys/include/frontends/ast/ast.h
/share/yosys/python3/__pycache__/smtio.cpython-35.opt-1.pyc
/share/yosys/python3/__pycache__/smtio.cpython-35.pyc
/share/yosys/python3/smtio.py
/share/yosys/xilinx/brams_bb.v
/share/yosys/xilinx/brams_init_16.vh
/share/yosys/xilinx/brams_init_18.vh
/share/yosys/xilinx/brams_init_32.vh
/share/yosys/xilinx/brams_init_36.vh
/share/yosys/xilinx/cells_xtra.v
/share/yosys/xilinx/drams.txt
/share/yosys/xilinx/drams_bb.v
/share/yosys/xilinx/drams_map.v

%changelog
* Mon Aug 29 2016 Benson Muite <benson_muite@emailplus.org>
- Initial version of package, make test fails at present with error needs iverilog from git