YosysHQ/yosys #219: opt_clean reverses assignments

animu5, 2016-08-29:
Here a minimal example: 
```verilog
module bsp_e(a, z);
	output z;
	input  a;
	wire s1;
assign
	s1 = ~a,
	z = s1;
endmodule
```

After executing opt_clean and write_verilog the output file looks like this:
```verilog
module bsp_e(a, z);
  input a;
  wire s1;
  output z;
  YNOT _0_ (
    .A(a),
    .Y(z)
  );
  assign s1 = z; /* this is reversed */
endmodule
```
This happens with commit b226893461af46f2183be8ca9dfab62b49133c71

cliffordwolf, 2016-08-29:
This is the intended behavior, and obviously a valid and useful transformation. Why do you think this is a problem?

animu5, 2016-08-29:
Seems like the I was wrong...
anyway, when using the above skript with the LGSynth91 benchmark circuits, specifically s13201.1_synth.v, s35932_orig.v, s35932_sweep.v, s35932_synth.v, outputs without drivers are created
for s1320.1_synth.v it's signal g1017 and g6207 and for the s35932 circuits its signal g16496
I thought the example above was the reason for this...

cliffordwolf, 2016-08-29:
I've downloaded LGSynth91.7z from http://ddd.fit.cvut.cz/prj/Benchmarks/, extracted 
Verilog/smlexamples/s13207.1_synth.v and looked at g1017. Here is the entire portion of the file that drives this ouput:

```
module \s13207.1.bench_excerpt(g1017);
  output g1017;
  wire \[894] ;
  reg g1029;

  assign g1017 = g1029;
  assign \[894]  = g1029;

  always begin
    g1029 = \[894] ;
  end

  initial begin
    g1029 = 0;
  end
endmodule
```

The only thing that's ok about that code is the initial block, which can simply ignored because of the always assignment to g1029. (I'm not omitting a `@*` here. This is how it been in the original s13207.1_synth.v file! However, there was no Verilog standard in '91, so maybe it was usual practice back then to use something like that in synthesizable code.)

So `g1017` is driven by `g1029` which is driven by `\[894]`, which is driven by `g1029`, ... This is just broken code. I see nothing wrong with the way Yosys handles that.

I'm now closing the issue. Feel free to post further comments if you find something that looks suspicious.

closed 2016-08-29 by cliffordwolf

cliffordwolf, 2016-08-29:
Even though I think that the behavior for this case is pretty much unspecified, I guess the expected behavior would be to not ignore the initial spec in favor of the "driver", because that driver is just a bunch of wires driving each other, and instead just set the whole thing to the init value. I've now implemented that in commit 4ea7054.