YosysHQ/yosys #224: Compile error

hungrymonkey, 2016-09-05:
I am writing support for my toy fpga. When I tried the sync my fork to the current master and compile it. I got this error. 

I guess the parser is unhappy with line 139

I am using opensuse 42.1 leap if that matters

make
[  0%] Building kernel/version_5415e0d.cc
[  0%] Building kernel/version_5415e0d.o
[  1%] Building kernel/driver.o
[  1%] Building techlibs/common/simlib_help.inc
[  1%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/register.o
[  2%] Building kernel/rtlil.o
[  2%] Building kernel/log.o
[  3%] Building kernel/calc.o
[  3%] Building kernel/yosys.o
[  4%] Building kernel/cellaigs.o
[  4%] Building kernel/celledges.o
[  5%] Building libs/bigint/BigIntegerAlgorithms.o
[  5%] Building libs/bigint/BigInteger.o
[  6%] Building libs/bigint/BigIntegerUtils.o
[  6%] Building libs/bigint/BigUnsigned.o
[  7%] Building libs/bigint/BigUnsignedInABase.o
[  7%] Building libs/sha1/sha1.o
[  8%] Building libs/subcircuit/subcircuit.o
[  9%] Building libs/ezsat/ezsat.o
[  9%] Building libs/ezsat/ezminisat.o
[ 10%] Building libs/minisat/Options.o
[ 10%] Building libs/minisat/SimpSolver.o
[ 11%] Building libs/minisat/Solver.o
[ 11%] Building libs/minisat/System.o
[ 12%] Building frontends/vhdl2verilog/vhdl2verilog.o
[ 12%] Building frontends/ilang/ilang_parser.tab.cc
[ 12%] Building frontends/ilang/ilang_parser.tab.o
[ 12%] Building frontends/ilang/ilang_lexer.cc
[ 13%] Building frontends/ilang/ilang_lexer.o
frontends/ilang/ilang_lexer.cc:912:35: warning: comparison of integers of different signs: 'int' and 'yy_size_t' (aka 'unsigned long')
      [-Wsign-compare]
                        for ( yyl = (yy_more_len); yyl < rtlil_frontend_ilang_yyleng; ++yyl )
                                                   ~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
frontends/ilang/ilang_lexer.cc:1926:17: warning: comparison of integers of different signs: 'int' and 'yy_size_t' (aka 'unsigned long')
      [-Wsign-compare]
        for ( i = 0; i < _yybytes_len; ++i )
                     ~ ^ ~~~~~~~~~~~~
2 warnings generated.
[ 13%] Building frontends/ilang/ilang_frontend.o
[ 13%] Building frontends/verilog/verilog_parser.tab.cc
frontends/verilog/verilog_parser.y:139.9-19: error: %define variable 'parse.error' is not used
frontends/verilog/Makefile.inc:8: recipe for target 'frontends/verilog/verilog_parser.tab.cc' failed
make: *** [frontends/verilog/verilog_parser.tab.cc] Error 1

hungrymonkey, 2016-09-05:
I forgot to mention that I did test it on the unmodified clifford's master branch. I apologize if the issue obvious but I guess I do not see a bug report report anywhere

cliffordwolf, 2016-09-05:
What output do you get for `bison --version`?

hungrymonkey, 2016-09-05:
bison --version
bison (GNU Bison) 2.7
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cliffordwolf, 2016-09-05:
I would assume that `%define parse.error verbose` is a feature new in bison 3.x. (bison 3.0 has been released in mid 2013. so it's not like it's been released yesterday. puzzling that opensuse is still on such an old version.)

You can simply remove this two lines from `frontends/verilog/verilog_parser.y`:

    %define parse.error verbose
    %define parse.lac full

That will give you slightly less verbose error reporting for syntax errors, but otherwise it should have no effect.

hungrymonkey, 2016-09-06:
ok, I will probably update the bison via the unstable repos. 

Thank you, I hope this problem will sort itself out with the next release of opensuse within a few months.

closed 2016-09-06 by hungrymonkey