YosysHQ/yosys #226: November+January: Upcoming freeze+FREEZE for Debian

smoe, 2016-09-08:
Hello,
Debian ships yosys and the whole pipeline will be in the next release. The nice development of its userbase casen be monitored [here](https://qa.debian.org/popcon.php?package=yosys). The next freeze is early November and an ultimate one early January (https://wiki.debian.org/DebianStretch). If there is an upcoming update of yosys or the downstream tools that you would prefer to see shipped instead of your current release then please kindly be aware of November, preferably.
Many thanks and regards,
Steffen

cliffordwolf, 2016-09-08:
Good to know. I plan to make the Yosys 0.7 release shortly after ORCONF, i.e. mid October. So that should work out just fine.

I'll leave this issue open until the release.

whitequark, 2016-10-18:
cc @azonenberg; maybe we should try to get a somewhat better gp4/cells_sim.v into yosys by this point?

smoe, 2016-10-18:
As instructed on https://www.reddit.com/r/yosys/comments/57xfgs/towards_yosys_07_please_test_now/ I have compiled the yosys head and it worked just fine on my little project. Somewhat stimulated by @whitequark, on my whishlist would be wand and wor (or a kick-but to go and implement it myself if that is feasible for mere mortals).

azonenberg, 2016-10-18:
@whitequark: I already have a few updates in my fork that aren't yet merged. I'll try and get cells_sim support for the remaining hard IP (even if we don't have back end toolchain support yet) asap.

cliffordwolf, 2016-11-03:
I have now released Yosys 0.7.
https://www.reddit.com/r/yosys/comments/5avscm/yosys_07/

closed 2016-11-03 by cliffordwolf

smoe, 2016-11-10:
Many thanks to @rubund for updating the yosys and berkeley-abc packages
https://packages.qa.debian.org/y/yosys.html