YosysHQ/yosys #227: torder -noautostop

Kmanfi, 2016-09-09:
Wondering why **torder -noautostop** sorts cells where 
cell $techmap\inst_m0.$memwr$\mem$memory_test.v:87$24
is above
cell $techmap\inst_m0.$procmux$35 
instead of
cell $techmap\inst_m0.$memrd$\mem$memory_test.v:84$23

Please run with
```
read_verilog memory_test.v
hierarchy -top memtest04

proc; opt; flatten; proc; opt; pmuxtree; opt; clean;
torder -noautostop

show
shell
```

[memory_test.v.zip](https://github.com/cliffordwolf/yosys/files/463364/memory_test.v.zip)

cliffordwolf, 2016-09-09:
The `torder` pass only looks at the topological ordering of the cells, not the cell semantics. It does not know that this memory read and write ports are related to each other.

The solution is to run `memory -nomap` to pack the memory read and write ports into one cell. For example like this:

    proc; opt; flatten; proc; opt; pmuxtree; opt; memory -nomap; clean;

Kmanfi, 2016-09-12:
With this:
`proc; opt; flatten; proc; opt; pmuxtree; opt; memory -nomap; clean;`

**torder -noautostop**
```
17. Executing TORDER pass (print cells in topological order).
module memtest04
  loop $procdff$54 $procdff$55 $procdff$56 $procmux$38
  cell $add$memory_test.v:14$2
  cell $add$memory_test.v:14$3
  cell $add$memory_test.v:14$4
  cell $procdff$55
  cell $procdff$56
  cell $procmux$38
  cell $procdff$54
  cell $techmap\inst_m0.$procmux$26
  cell $techmap\inst_m0.$procmux$29
  cell $techmap\inst_m0.$procmux$32
  cell inst_m0.mem
  cell $techmap\inst_m0.$procdff$49
```

It would help me **a lot** if cells were able to order by depth of logic level.
Then cell $procdff$55 would be right above $mem, in same logic depth as cells
 cell $techmap\inst_m0.$procmux$26
 cell $techmap\inst_m0.$procmux$29
 cell $techmap\inst_m0.$procmux$32
 
**Feature request: Option to order cells topological by logic depth.**

cliffordwolf, 2016-09-12:
> It would help me a lot if cells were able to order by depth of logic level.

Can you elaborate a bit on what you are trying to accomplish? I'm asking because:

> `loop $procdff$54 $procdff$55 $procdff$56 $procmux$38`

There is no topological ordering for this netlist with `-noautostop` . It contains this loop.

Kmanfi, 2016-09-12:
This request is related to custom back end written for Yosys. I use the output of the back end for some Python project. Now big number of projects work but there are some that don't. Bugs are now related to reg to reg delay line type of connections where there is no cell between DFF's.
I tried insbuff but it doesn't add buffs between directly connected sequential cells like $mem driving $dff. Insbuf will also split net to single bit, so it seems there is no support multibit $buf.

So I have two tracks. Either guarantee there is always cell between directly connected sequential cells OR order sequential cells as starting from module outputs towards module inputs.
Why this ordering is important. Reason is that the Python project relies on ordering where logic depth of sequential cells is known.
Currently the back end exports torder result as part of exported data and the Python project collects sequential and non sequential cells for further processing.

cliffordwolf, 2016-09-12:
So why are you using `-noautostop`? I'm not sure if I understood everything you said, but I see no indication that `-noautostop` would be useful since there is no indication that you are limiting yourself to circuits that are free of topological loops unless FFs are considered not to be part of the graph (which is the behavior without `-noautostop`).

> I tried insbuff but it doesn't add buffs between directly connected sequential cells [..]

That's not what insbuf does. Quoting the insbuf help message:

> Insert buffer cells into the design for directly connected wires.

An example for that would be:

```
module test(input in, output out);
  assign out = in;
endmodule
```

A wire connecting two cells (FFs or otherwise) is the direct opposite of "directly connected wires", i.e. wires connected to each other without any cells inbetween.

> Either guarantee there is always cell between directly connected sequential cells [..]

Imo this sounds like you have a fundamental design bug in the code that processes the netlist if this is a requirement for the generated netlist.

Why not simply treat each FF in your python code like an FF followed by a buffer, if this is really necessary to get synchronization right in your Python code? To me that sounds like a better solution than working-around that bug by adding dummy buffer cells during the netlist generation step, i.e. a completely different part of your system.

> [..] OR order sequential cells as starting from module outputs towards module inputs.

Such an ordering generally does not exist. It certainly does not exist in the example you have posted, thus the `loop $procdff$54 $procdff$55 $procdff$56 $procmux$38` message.

Kmanfi, 2016-09-13:
Thank you for your comments. I have to rethink this more...

closed 2016-09-13 by Kmanfi