YosysHQ/yosys #229: Regression test fails on Fedora aarch64

brouhaha, 2016-09-14:
I'm packaging yosys for Fedora, and the package builds and passes all regression tests on x86 (both 32-bit and 64-bit), and 32-bit ARM, but one of the realmath tests fails on 64-bit ARM.  The build log is at:
    https://gist.github.com/brouhaha/605f6b47ccb505369f5a1fb06a544d84

As part of packaging for Fedora, I moved the includes from /usr/share/yosys/include to /usr/include/yosys, with the corresponding changes to yosys-config, and used the Fedora-required CXXFLAGS:
    -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic

It seems entirely possible that the fault is with something else rather than with yosys, but I don't yet know. The Fedora Rawhide buildroot currently has an iverilog git snapshot from August 2015:
   https://github.com/steveicarus/iverilog/commit/6d0ab9978f036e6029858e0d1b0bdab52e3fbad7
I can't change the iverlog in the buildroot without coordination from Fedora's iverilog package maintainers, so I'll try building with a newer iverilog snapshot on a system of my own, a Rasberry Pi 3 running Fedora 24 aarch64.

cliffordwolf, 2016-09-14:
The realmath tests are randomly auto-generated. Without the generated file that failed there is nothing I can do. I think this has nothing to do with 64-bit ARM, you just managed to hit a random test case that fails. With 99% certainty this is just a bad test case. Comparing two implementations of floating point arithmetic is unfortunately prone to false errors..

I run "make test" easily 500-1000 times a year and I hit a bad realmath test case maybe once every year. I always check, but it always turns out that both results are OK.

If you can get the failing test case (e.g. just zip the entire `tests/realmath/temp/` directory), I'd be happy to look into it. Otherwise I'd recommend to simply re-run the build on 64-bit ARM. It should work just fine.

brouhaha, 2016-09-15:
Confirmed that the problem was a randomly generated test.  I'm patching things to allow for optional use of a specified seed value. I'll open a new issue with a proposed patch.

closed 2016-09-15 by brouhaha