YosysHQ/yosys #232: liberty-files

leviathanch, 2016-09-17:
There is an issue with Yosys when trying to load the liberty-files in order to use the blackbox-model for the sram and flash with `"read_liberty -lib -ignore_miss_dir -setattr blackbox"`
Apparently **".bus"** is not parsed, only **".pin",** which makes it impossible to wire the address bus and data bus because A, D and Q are busses.

cliffordwolf, 2016-09-17:
Do you have an example .lib file that I can use for testing? (Send it to my private email if you prefer it not to be in the public.)

leviathanch, 2016-09-18:
Yes and ok!

cliffordwolf, 2016-09-18:
Commit 13a03b8 adds just enough support to fix the example you have sent me. Just let me know if you find more examples that would require additional work to be done.

closed 2016-09-18 by cliffordwolf