YosysHQ/yosys #234: for loops take long for yosys to compile

smoe, 2016-09-19:
Hello,

I made an observation on the time yosys invests on my code. This is not about the time the FPGA needs to execute it.

The implementation of for loops is apparently not in a way that the typical assembly/C programmer expects. The _extra time_ (and memory) yosys needs rewrite a for-begin-end construct is not a constant but it depends on the number of lines embraced by the loop and, in particular so, on number of iterations. And that dependency is considerably worse than linear.

Example:
```
module takeslong (input iCE_CLK);
	reg [220:0] i;
	reg v=0;
	reg [5:0] w=0;
	always @( posedge iCE_CLK ) begin
		for( i=0; i<50000; i=i+1) begin
			if (v) begin
				v <= ~v;
				w <= w+1;
			end
		end
	end
endmodule
```
The above featuring 50000 iterations does not complete the RTLIL representation stage after 5 minutes, with one or two lines. With 10000 iterations and only the v line it takes 8 seconds, with 10000 and both lines it is close to 20 seconds.  1000 iterations are instantaneous.

I presume I could somehow work around using the for loop in the first place, but this iCEstick will attract more and more typical IT folks that will enjoy, say, implementing something like the https://en.wikipedia.org/wiki/Sieve_of_Eratosthenes and they were all raised/brain-washed with for loops. My heart beats for all those folks interseted in application acceleration who are much alike me very much prone to fall into this trap. Can they be helped? Is this a bug/dysfeature of yosys? Or if not, would it be possible for yosys to warn about an inappropriate use of the for loop rather than just eating up all system resources (>100GB of memory on my main project that is problem-freely synthesising with ISE)?

Many thanks and regards,

Steffen

cliffordwolf, 2016-09-19:
> [..] 50000 iterations [..]  iCEstick will attract more and more [..]

Your example describes a circuit with 350k LUTs (50k 6-bit adders = 300k LUTs, 50k 1-bit inverters = 50k LUTs). Not really something that would fit on an iCEstick (which has 1k LUTs).

Of course your example will in the end produce an empty circuit because all of that stuff is going to be optimized away. But that is of course not a real use case. When you have real code then each iteration would create some kind of circuit. Even if it would only be one LUT per iteration, it would still be much much larger than anything that could fit on an iCE40 FPGA.

> The implementation of for loops is apparently not in a way that the typical assembly/C programmer expects

That's what ALL synthesis tools do. They HAVE to unroll all loops. What else would you expect?

I'm closing this as a wontfix. There is probably a performance bug somewhere in the unrolling that has O(n^2) runtime. But fixing this won't change the amount of memory required to store the unrolled AST and/or generated RTLIL circuit representation. Feel free to dig into this and open a pull request with a fix if you find something worth fixing.

closed 2016-09-19 by cliffordwolf

smoe, 2016-09-21:
Thank you for your time and kind reply.
> That's what ALL synthesis tools do. They HAVE to unroll all loops. What else would you expect?
The generate precompile instructions I expected to unroll, the for loops I expected to implement a particularly clever finite state machine. This would then also help the interpretation of variable conditions.

cliffordwolf, 2016-09-21:
>  the for loops I expected to implement a particularly clever finite state machine.

Can you give an example what that would look like? I.e. a short Verilog example using a for loop, and a corresponding equivalent Verilog example using the clever FSM you'd like the synthesis tool to infer?

(I'm pretty sure that while trying to produce such an example you will realize that this is not possible. But it would be very exciting if I'm wrong and might even have major implications for mathematical fields like complexity theory. :)

Usually the amount of computation you can do is proportional to the time you have to do it and by the size of the circuit you can create. Your example is doing 50000 "operations" in one cycle. That means you'd have to have 50000 instances of the circuit that performs the operation. You could of course do the same work in 50000 cycles using only one instance. But that would not be equivalent to the original Verilog code anymore.

smoe, 2016-09-21:
You have answered it to the point that I rest assured you do not want me to provide the FSM any more that would not match the 1-cycle for-loop's semantics you just explained. My mistake was to have just trusted that for loop to take as many circles as it needs, 50000 most likely. And  I would not have minded a partial unrolling to speed things up and fill the array. IIRC, I was just trained by those few lines here http://www.asic-world.com/verilog/verilog_one_day2.html . I was a bit amused when I now reread this and it clearly says "works like C/C++".

What strikes me, still, is that it was so difficult for me to come up with a test case.  My loop length is typically more at 8 (eight), not those nifty 50000. Commenting out different parts of the code nonetheless clearly identified the for loops as a culprit. 

My application btw. uses a hash table to record the presence/absence of a feature. That feature is an integer distance, so for any new object I iterate in a for loop through all prior accepted objects and compute distances. If any such distance is already existing, I drop the object. With ISE this is ia a "for (i=0;stillgood && i<n;i=i+1) stillgood= ~hash[newobject - object[i] ]" kind of loop and I see now why yosys does not allow this. 

To use the iCEstick together with yosys is key for my ambitions. I will hence go on and substitute the for loops with FSMs. Anyway, should there be any value in the original code for you to help the resilience of yosys, it is here https://github.com/smoe/optimal-golomb-ruler-verilog . The code synthesizes for the small Xilinx Spartan6 with ISE but admittedly needs some overhaul, still, to be recognizable by folks that know how to develop in Verilog, properly. But, well, this may be why it is of some value for you, possibly :-)

smoe, 2016-09-21:
I found http://embdev.net/topic/for-loop-in-verilog-code which kind of confirms that noobs run into such issues. One comment was that beginners should just not be allowed to use any loop constructs. Sounds like right.

dawidgrabowski, 2016-09-21:
@smoe Besides the fact that whatever you did in the for loop doesn't really make any sense (but that's I guess because you are not familiar with how the loops work in Verilog) I can give a small advice how you can decrease memory usage and runtime when you need to use large loops. If you have following loop:

```
for( i=0; i<50000; i=i+1) begin
    ...
end
```

try to break it to smaller loops i.e.:

```
for( i=0; i<=24999; i=i+1) begin
    ....
end
for( i=25000; i<50000; i=i+1) begin
    ....
end

```