YosysHQ/yosys #237: Opt, clean does not remove unnecessary assigns after flatten

Kmanfi, 2016-09-23:
Opt after flatten does not remove unnecessary assigns like below.
```
  assign \inst_m0.ADR_RD  = delay_line1;
  assign \inst_m0.ADR_WR  = delay_line2;
  assign \inst_m0.DATA_IN  = sum_out[3:0];
  assign \inst_m0.RE  = wr_enable;
  assign \inst_m0.WE  = sum_out[0];
  assign \inst_m0.clk  = clk;
```

Please run:
```
read_verilog memory_test.v
hierarchy -top memtest04
proc; opt; flatten; opt; clean
write_verilog -noattr -noexpr fooooo.v
```

[memory_test.v.zip](https://github.com/cliffordwolf/yosys/files/489148/memory_test.v.zip)

cliffordwolf, 2016-09-23:
Without having looked at the .zip file, I think you are looking for `opt -purge` (or simply `clean -purge` or `opt_clean -purge`). By default, `opt` preserves all original net names, if possible. This significantly improves the quality of VCD files produced by post-synthesis simulation and simplifies analyzing issues with the synthesis flow.

Kmanfi, 2016-09-23:
-purge is exactly what I need in my backend. Thank you.

closed 2016-09-23 by Kmanfi