YosysHQ/yosys #240: PATH_MAX not defined on GNU HURD

smoe, 2016-10-06:
Hello,

this is about yosys not compiling for the hurd platform at https://github.com/cliffordwolf/yosys/blob/master/kernel/yosys.cc#L653 . Ok, there will not be many users of yosys on hurd in the first place, admittedly. Hurd does not know about maximal path lengths and as such does not define any such limit. Others have run into this before, as in
https://github.com/google/protobuf/issues/2000
and the porting guidelines are insisting on that one needs test for PATH_MAX to be defined prior to using it
https://www.gnu.org/software/hurd/hurd/porting/guidelines.html#PATH_MAX_tt_MAX_PATH_tt_MAXPATHL

Do you feel like adding something like
 #include <limits.h>
+#if defined(__GNU__)
+#define PATH_MAX 4096
+#endif
for immediate compatibility with HURD? And maybe the ifdef in line 650 should explicitly mention __GNU__, too.

It could possibly come up with a requestive pull request if this is any help.

Many thanks and greetings

Steffen

smoe, 2016-10-06:
I just found an interesting sniplet in "man 2 readlink" which may substitute your use of PATH_MAX with a completely dynamic implementation - shall I prepare a respective patch?

smoe, 2016-10-08:
I applied the suggestions from the above mentioned man page to readlink(2) which looks like this: ```
diff --git a/kernel/yosys.cc b/kernel/yosys.cc
index 08fee97..99373b2 100644
--- a/kernel/yosys.cc
+++ b/kernel/yosys.cc
@@ -647,17 +647,40 @@ struct TclPass : public Pass {
 } TclPass;
 #endif
 
-#if defined(__linux__) || defined(__CYGWIN__)
+#if defined(__linux__) || defined(__GNU__) || defined(__CYGWIN__)
 std::string proc_self_dirname()
 {
-       char path[PATH_MAX];
-       ssize_t buflen = readlink("/proc/self/exe", path, sizeof(path));
+
+       struct stat sb;
+       if (-1 == lstat("/proc/self/exe", &sb)) {
+               log_error("lstat(\"/proc/self/exe\") failed: %s\n", strerror(errno));
+               exit(-1);
+       }
+
+        ssize_t allocated=(0==sb.st_size?4096:(sb.st_size + 1));
+
+       char *path = (char *) malloc(allocated);
+       if (path == NULL) {
+               log_error("proc_self_dirname: insufficient memory\n");
+               exit(-1);
+       }
+
+       ssize_t buflen = readlink("/proc/self/exe", path, allocated);
        if (buflen < 0) {
-               log_error("readlink(\"/proc/self/exe\") failed: %s\n", strerror(errno));
+               log_error("proc_self_dirname: readlink(\"/proc/self/exe\") failed: %s\n", strerror(errno));
        }
+       if (buflen > allocated) {
+               log_error("proc_self_dirname: size of symlink exceeds memory allocation from lstat() (== %ld) and readlink() (== %ld)\n",buflen,sb.st_size);
+               exit(-1);
+               }
+       path[buflen] = '\0';
+
        while (buflen > 0 && path[buflen-1] != '/')
                buflen--;
-       return std::string(path, buflen);
+
+        std::string ret = std::string(path,buflen);
+       free(path);
+       return ret;
 }
 #elif defined(__APPLE__)
 std::string proc_self_dirname()
```
The annoying part is that the proc fs apparently does not give any size information for the link, as seen in ```
$ ls -l /proc/self/exe
lrwxrwxrwx 1 smoe smoe 0 Oct  8 18:42 /proc/self/exe -> /bin/ls
```
It reads "0", seriously. Hence, above fancy code same as ignorantly just goes off and occupies that 4096 bytes, i.e. no better than the simple #define in my reading. Should this be fixed in /proc?

cliffordwolf, 2016-10-11:
Thanks for reporting. Commit 59508c9 should fix this issue.

closed 2016-10-11 by cliffordwolf