YosysHQ/yosys #242: Compiling on MacOS 10.12

davidgraham, 2016-10-10:
While compiling Yosys on MacOS 10.12, I received the following errer:
``
frontends/ast/dpicall.cc:25:10: fatal error: 'ffi.h' file not found
``

I added "_-L/usr/local/opt/libffi/lib_" to the Makefile under the Darwin specific flags

``
LDFLAGS += -L/opt/local/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/libffi/lib
``

and compilation completed, passing all tests

cliffordwolf, 2016-10-12:
hmm.. Why does adding a `-L` option fix a not found include file? Why do the ffi `pkg-config` commands not work in this case? I don't have a Mac, so I cannot test this.

maknoll, 2016-10-12:
TLDR; Build worked fine on my 10.12 machine, OPs pkg-config might be screwed up.

The average Developers Mac has multiple installations of pkg-config. E.g. there is a default Mac OS one, there might be one from the Homebrew package manager and maybe another one from Macports, like OP seems to have. Depending on how well the software using pkg-config is written it might uses hardcoded paths and ignores environment vars.
So OP might has libffi installed via macports, but his PKG_CONFIG_PATH variable might point elsewhere. Also his fix wouldn't work on my machine because i do not use Macports wich keeps stuff in /opt/ (which is sacrilege IMHO) but Homebrew, wich keeps stuff in /usr/local/.

Installing 10.12 might has screwed up something else, i'd recommend running 

    xcode-select --install

just in case.

adhorden, 2016-10-12:
I have an idea why this is. I will put together a PR 😬.

davidgraham, 2016-10-12:
Apologies, the error is on my system.  After upgrading MacOS and Xcode, I thought I'd ran xcode-select after the upgrade (I'd even wiped and re-installed homebrew and it's packages without issue; no macports), but I ran it again and was able to compile without the above change.
Thanks @maknoll!

closed 2016-10-12 by davidgraham