YosysHQ/yosys #247: Segfault when using recursive functions

AlexDaniel, 2016-10-16:
Here is a golfed example from a larger description I was writing:
```verilog
// segfault.v
module segfault(output y);
   function get_xxx;
      input i;
      begin
           get_xxx = i == 0 ? 1 : get_xxx(0);
      end
   endfunction
   assign y = get_xxx(0);
endmodule
```

As I see it, there is enough information during compilation to unwrap this function. In fact, in this case recursion does not happen at all (it should return ``1`` immediately without reentering ``get_xxx`` function).

I run it like this:
```
yosys -p 'prep' segfault.v
```

Here is the output:
```
 Yosys 0.6 (git sha1 5869d26, gcc 6.2.0-4 -O2 -fdebug-prefix-map=/build/yosys-aIBmDV/yosys-0.6=. -fstack-protector-strong -fPIC -Os)

-- Parsing `segfault.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `segfault.v' to AST representation.
Generating RTLIL representation for module `\segfault'.
Segmentation fault
```

Am I doing something incorrectly? If so, maybe it is possible to implement a better failure mode for this case?

AlexDaniel, 2017-07-01:
Now I wonder if this example is correct. Shouldn't it be marked as [``automatic``](https://stackoverflow.com/questions/35170/verilog-automatic-task)?

cliffordwolf, 2017-07-03:
Since the function has no local variables, `automatic` shouldn't make any difference afaik. But yes, in general a recursive function should be declared `automatic`. However, in synthesis all functions are automatic anyways. (Automatic functions are slower to simulate, that's why not all functions are "automatic" by default afaik.)

tilk, 2020-04-05:
One of my students recently tried to use recursive functions in DigitalJS, and got a segmentation fault. I imagine this is the same bug as this one. This is the code in question, with irrevelant parts removed:
```verilog
module bar(output o);
   function automatic foo(integer k);
      if (k == 0)
        foo = 1'b0;
      else
        foo = foo(k-1);
   endfunction // foo
   assign o = foo(0);
endmodule
```

zachjs, 2021-01-01:
With #2517 and #2518, @tilk's example above should now be working correctly. @AlexDaniel's original example _nearly_ works, but the use of the ternary rather than a traditional branch still results in infinite recursion and a segfault. I will look into that separately.

whitequark, 2021-01-01:
Very nice, thank you!

PR #2586 (2021-02-21): verilog: support recursive functions using ternary expressions
This adds a mechanism for marking certain portions of elaboration as occurring within unevaluated ternary branches. To enable elaboration of the overall ternary, this also adds width detection for these unelaborated function calls.

While this implementation is meant to be low-impact for prior inputs, I do think the introduction of `unevaluated_tern_branch` is not necessarily ideal. I'm eager for any feedback or alternative suggestions.

This fixes #247 (the original example uses a ternary in this way).
closed 2021-02-21 by whitequark