YosysHQ/yosys #249: Feature request: Detect incorrect "assign" to a synchronously assigned "reg" type

azonenberg, 2016-10-19:
Consider the following code:

`
reg foobar = 0;
always @(posedge clk) begin
    foobar <= some_complex_expression_here;
end

assign foobar = 1;
`

Yosys will silently ignore the "assign" statement. This should probably result in a "multiple drivers" error instead: A given identifier can be either combinatorially or synchronously assigned, but not both.

yugr, 2016-10-19:
It shouldn't be possible to continuously assign reg to begin with...

cliffordwolf, 2016-10-19:
Quoting `help read_verilog`:

> Note that the Verilog frontend does a pretty good job of processing valid
verilog input, but has not very good error reporting. It generally is
recommended to use a simulator (for example Icarus Verilog) for checking
the syntax of the code, rather than to rely on read_verilog for that.

A design flow in which one does not simulate a design is fundamentally broken anyways.

> It shouldn't be possible to continuously assign reg to begin with...

correct. For the most part the yosys verilog front-end doesn't care if a signal is declared `reg` or `wire`. (Verilator does the same, btw.) But most simulators will catch that error. For example icarus verilog:

    test.v:7: error: reg foobar; cannot be driven by primitives or continuous assignment.

> This should probably result in a "multiple drivers" error instead.

Multiple drivers is not an error in Verilog per-se, as verilog does support tri-state logic.

closed 2016-10-19 by cliffordwolf