YosysHQ/yosys #253: fsm encoding

yuanxue001, 2016-10-25:
So i just try a quick example of following scripts:

read_verilog test.v
proc; opt; fsm -nomap
fsm_export -o test.kiss2

I got the state encoding assignments, and it seems the states are encoded by one-hot.
The manual said yosys supports one-hot and binary encoding.  I tried to do binary encoding, no success.
How can I use command "fsm_recode" to change encoding strategy, can you just give a quick guide? Also if I want to implement other encoding strategy, can you give me a brief hint of how to do that, or it is impossible? Thx

cliffordwolf, 2016-10-25:
Simply call `fsm` with the `-encoding` option, or set the `fsm_encoding` attribute on the state signal. Possible values are `auto`, `one-hot` and `binary`. For example:

```
read_verilog test.v
proc; opt; fsm -encoding binary -nomap
fsm_export -o test.kiss2
```

See `passes/fsm/fsm_recode.cc` for the implementation of FSM recoding. This is the place where you'd add new encoding types.

yuanxue001, 2016-10-25:
Thank you, I tried both one-hot and binary encoding, I thought different fsm encoding strategies would cause different cell costs, such as use different numbers of flip-flops and logic gates. But when I use "stat" to collect these data, it seems that encoding strategies did not affect logic at all.
Can you give a brief answer about this?  Thank you so much, clifford.

cliffordwolf, 2016-10-25:
You don't seem to map your circuit down to a gate level representation. So in both cases the state register gets mapped to a single `$dff` cell, just with different `WIDTH` parameters.

If this doesn't explain it: Can you post the `test.v` and complete yosys scripts you are using?

yuanxue001, 2016-10-25:
my test.v is as following:
//////////////////////////////////////////////////////////////////////////////////////////
module s27(input clk,input[3:0] in, output [7:0] out); 
        reg[2:0] state;
        always @(posedge clk) begin
                out<=0;
                case (state)
                        0: begin
                                if(in == 0) begin
                                        state<=1;
                                        out<=0;
                                end
                                if(in == 1) begin
                                        state<=2;
                                        out<=1;
                                end
                                if(in == 2) begin
                                        state<=5;
                                        out<=2;
                                end
                        end
                        1: begin
                                if(in == 0) begin
                                        state<=1;
                                        out<=3;
                                end
                                if(in == 1) begin
                                        state<=3;
                                        out<=4;
                                end
                                if(in == 2) begin
                                        state<=1;
                                        out<=5;
                                end
                                if(in == 3) begin
                                        state<=3;
                                        out<=6;
                                end
                                if(in == 4) begin
                                        state<=2;
                                        out<=7;
                                end
                                if(in == 5) begin
                                        state<=5;
                                        out<=8;
                                end
                                if(in == 6) begin
                                        state<=3;
                                        out<=9;
                                end
                                if(in == 7) begin
                                        state<=5;
                                        out<=10;
                                end
                                if(in == 8) begin
                                        state<=3;
                                        out<=11;
                                end
                                if(in == 9) begin
                                        state<=2;
                                        out<=12;
                                end
                        end
                        2: begin
  if(in == 0) begin
                                        state<=2;
                                        out<=13;
                                end
                                if(in == 1) begin
                                        state<=2;
[... 155 lines cut ...]
                                end
                                if(in == 9) begin
                                        state<=2;
                                        out<=52;
                                end
                        end
                endcase
        end
endmodule
//////////////////////////////////////////////////////////////////////////////////////////
I used ISCAS89 benchmark directly and it did not work, so I modified one small benchmark in a way yosys can read.

I use scripts below:

**read_verilog test.v
proc; opt; fsm -encoding binary -nomap
techmap; stat**

then I run another round with different encoding:

**read_verilog test.v
proc; opt; fsm -encoding one-hot -nomap
techmap; stat**

I have to keep "-nomap" ,if I don't the binary case will just quit as error. And the stat gives me the exact the same cell numbers and wire bits. I expect as least the $_DFF_P_ should be different.  Please take a look, I'm not sure how I was wrong.

cliffordwolf, 2016-10-25:
> I have to keep "-nomap" ,if I don't the binary case will just quit as error.

This is now fixed in commit d9d38ee.

The problem was that this FSM does not have a reset mechanism. (I.e. in simulation it would simply be stuck in the `3'bxxx` state.) Such FSMs are very unusual input to a synthesis tool.. :)

> And the stat gives me the exact the same cell numbers and wire bits.

Because the FSM is still just a single $fsm cell. Use `-nomap` with current git head, and add at least `opt` after techmap, and you'll get the results you'd expect:

```
read_verilog test.v
proc; opt; fsm -encoding binary
techmap; opt; stat
```

and

```
read_verilog test.v
proc; opt; fsm -encoding one-hot
techmap; opt; stat
```

produce the following stat outputs:

```
==> test1.log <==
     $_AND_                         47
     $_DFF_P_                        9
     $_MUX_                         93
     $_NOT_                         29
     $_OR_                         135
     $_XOR_                         17

==> test2.log <==
     $_AND_                         47
     $_DFF_P_                       11
     $_MUX_                         90
     $_NOT_                         24
     $_OR_                         119
     $_XOR_                         14
```

And using `opt; abc; opt` instead of simply `opt`:

```
==> test1.log <==
     $_AND_                         20
     $_AOI3_                         8
     $_DFF_P_                        9
     $_MUX_                          4
     $_NAND_                         7
     $_NOR_                          5
     $_NOT_                         10
     $_OAI3_                         2
     $_OAI4_                        10
     $_OR_                          37
     $_XOR_                          1

==> test2.log <==
     $_AND_                         15
     $_AOI3_                         7
     $_AOI4_                         3
     $_DFF_P_                       11
     $_MUX_                          5
     $_NAND_                         5
     $_NOR_                         13
     $_NOT_                          9
     $_OAI3_                         9
     $_OAI4_                         3
     $_OR_                          21
     $_XOR_                          2
```

closed 2016-10-25 by cliffordwolf