YosysHQ/yosys #257: Displayed FSM output values are not very useful

AlexDaniel, 2016-11-02:
From one of [your answers](http://stackoverflow.com/a/32648163/2698019):

> Note: The FSM outputs in this case are not directly the four ``O`` signal bits. Instead Yosys created an FSM with a three bit output and an encoder outside of the FSM for creating the four ``O`` signal bits.

That's OK, but when exporting a FSM as a kiss2 file I am expecting to see the actual outputs. Do I get it right that currently there is no way to do that? And if not, then what's the idea of having a kiss2 file with these half-backed output values?

cliffordwolf, 2016-11-02:
The `fsm_expand` pass can be used to merge extra logic into the FSM. I've now added some minor fixes and improvements to this mechanism in commit 0c8e973. Update to latest git head an simply call `fsm` with the option `-fullexpand`.

In the example from the stack overflow example this will produce an FSM with the 4 output signal (before the final FF stage that produces the `O` signals), instead of the 3 intermediate signals generated by the FSM without the `-fullexpand` option.

closed 2016-11-02 by cliffordwolf

AlexDaniel, 2016-11-02:
That was quick. Thank you very much!