YosysHQ/yosys #259: Less than awesome warnings about init values in an FSM

AlexDaniel, 2016-11-02:
There are two warnings I don't understand:

> Init value on fsm state registers are ignored. Possible simulation-synthesis mismatch!

Does it mean I cannot have an initial state in an FSM? If so, why? And what kind of mismatch does it refer to? Perhaps this warning can be expanded a little bit, I don't think it will be a problem if the warning is more than one sentence.

> FSM seems to be self-resetting. Possible simulation-synthesis mismatch!

Again, sounds like something I should not do. Why, and what kind of mismatch should I expect?

cliffordwolf, 2016-11-02:
It would help if you'd posted some code, then I could tell you how to improve it.

The messages really say it all:

> Init value on fsm state registers are ignored. Possible simulation-synthesis mismatch!

init values on fsm state registers are ignored. I don't know what else I could add to make what is going on. they are ignored. I.e. the design is treated like the initialization is not present.

It should be pretty obvious that the design with and without the initialization would simulate differently. So there is a mismatch espected between simulation of the original design, and post-synthesis simulation. Again, I don't know what else I could add to make this more clear.

> FSM seems to be self-resetting. Possible simulation-synthesis mismatch!

A self-resetting FSM might not be self-resetting anymore after recoding. E.g. it could be that in the original (dense) encoding there are no invalid states. But in the new (one-hot) encoding there are invalid states. This, again, might cause a possible simulation-synthesis mismatch.

Usually an FSM is not extracted automatically if one of those things is true. However, if you force FSM extraction by setting the `fsm_encoding` attribute then you will get those warnings.

Closing this issue now. I'll reopen it if it becomes actionable again, e.g. because additional information is provided.

closed 2016-11-02 by cliffordwolf

AlexDaniel, 2016-11-03:
OK, after reading https://github.com/cliffordwolf/yosys/issues/188 I think I get it now. I am not sure, maybe the warning should actually say that init values are not yet supported, that would probably make more sense.