YosysHQ/yosys #260: Hiding temp variables from “show” output

AlexDaniel, 2016-11-03:
Let's say we have this module:

```verilog
module padder (input [N-1:0] a, input [N-1:0] b, input c_in,
               output reg [N-1:0] s, output reg c_out);
   parameter N = 4;

   genvar i;
   genvar j;
   reg [N-1:0] p;
   reg [N-1:0] g;
   reg [N  :0] c;
   for (i = 0; i < N; i = i + 1) begin
      assign p[i] = a[i] | b[i];
      assign g[i] = a[i] & b[i];
   end
   assign c[0] = c_in;
   for (i = 0; i < N; i = i + 1) begin
      reg [i:0] temp;
      assign temp[0] = &{p[i:0], c_in};
      for (j = 1; j <= i; j = j + 1)
        assign temp[j] = &{g[j-1], p[i:j]};
      assign c[i+1] = |{temp, g[i]};
   end
   for (i = 0; i < N; i = i + 1)
     assign s[i] = ^{a[i], b[i], c[i]};
   assign c_out = c[N];
endmodule
```

Now, if we load this module and run ``show``, we will get this:

[image]

Which is alright, but I want to hide genblocks (they are there just to aggregate the values, so I am not interested in seeing them on the graph). That is, I want ``reduce_and``-s to be connected directly to ``reduce_or``-s, without passing through ``$genblock$padder.v:*.temp``.

After reading [Yosys AppNote 011](http://www.clifford.at/yosys/files/yosys_appnote_011_design_investigation.pdf) my expectation was that it should be possible (given how flexible the whole thing is).

I tried this:
```
show -colors 42 padder/* padder/*genblock* %d
```

But the result is not what I wanted:
[image]

So is it just me being unable to figure out how to do that, or is this kind of thing missing?

cliffordwolf, 2016-11-04:
Do you want to hide only the netnames or also the nets and/or circuitry associated with it?

That the netnames are visible is a bug that I have now fixed in commit 2874914. Either update to git head or, as a work-around, run `rename -hide n:$genblock$*` before running `show`.

AlexDaniel, 2016-11-04:
Alright, that's better:
[image]

However, I'm not sure if I like this kind of thing:
[image]

If that's what you mean by “nets and/or circuitry associated with it”, then yes. Basically, I just think that the way it groups the signals is not very helpful in this particular case.

cliffordwolf, 2016-11-04:
> However, I'm not sure if I like this kind of thing

What's your Yosys script? Looks like you are not even running `opt_clean`. I'd use at least something like `prep; show -colors 42`. The output of `prep; splitnets -ports;; show -colors 42` is probably even better considering the bit-wise nature of that design.

AlexDaniel, 2016-11-04:
The thing I mentioned remains after ``opt_clean``. Same thing with ``prep`` (yes, I usually run ``prep``, but as it didn't change anything regarding this signal grouping I didn't want to introduce any extra commands for reproducing the issue). Yes, ``splitnets -ports`` is an interesting idea, but that's not what I wanted.

After looking at it again I am not longer sure if there is any issue. Yes, the way it looks is a bit weird, but it is probably OK. Feel free to close this issue if you think that there is no problem.

closed 2016-11-04 by cliffordwolf