YosysHQ/yosys #262: SystemVerilog “logic” type seems to be supported

AlexDaniel, 2016-11-05:
[README](https://github.com/cliffordwolf/yosys/blob/master/README) file has a section on “Non-standard or SystemVerilog features”. As far as I can see, ``logic`` is actually supported, but it is not listed in the README.

If you try to use ``logic`` type, you will get this error message:
```
Lexer warning: The SystemVerilog keyword `logic' (at foo.v:2) is not recognized unless read_verilog is called with -sv!
```
Which suggests that ``logic`` is actually supported already. If it is so (and if it is known that it is mostly bug-free), then it should be listed in the README.

cliffordwolf, 2016-11-05:
Quoting the README file:

> The keywords "always_comb", "always_ff" and "always_latch", "logic" and
  "bit" are supported.

closed 2016-11-05 by cliffordwolf

AlexDaniel, 2016-11-05:
Indeed! Sorry :)