YosysHQ/yosys #264: Post-synthesis simulation, Xilinx

AlexDaniel, 2016-11-05:
[This answer](http://stackoverflow.com/a/35939285/2698019) is very promising. However, I need the same thing for Xilinx FPGAs. So I tried this:
```bash
yosys -p 'synth_xilinx -top test; write_blif test.blif' test.v
yosys -o test_syn.v test.blif
iverilog -o test_post -D POST_SYNTHESIS test_tb.v test_syn.v \
                        `yosys-config --datdir/xilinx/cells_sim.v`
./test_post
```

But I am not getting the right output:
```
xxxx
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
0000
```

Looking at the ``test_syn.v`` file, it seems like LUTs are not initialized.

cliffordwolf, 2016-11-05:
If you want a Verilog output file, just create one:

```
yosys -p 'synth_xilinx -top test; write_verilog test_syn.v' test.v
```

The only reason we go thru `.blif` in the example on SO is because this is the file used as input to archne-pnr in the iCE40 flow. There is no reason to do that in the Xilinx flow.

If you really want to use .blif, make sure to include cell parameters by calling `write_blif` with `-param`. This uses the non-standard BLIF statement .param in the generated file. (there is no way of including cell parameters in standard conforming BLIF, so if you just run `write_blif` then all cell parameters are ignored.)

closed 2016-11-05 by cliffordwolf

AlexDaniel, 2016-11-05:
Oh! That makes sense! Thank you very much.

cliffordwolf, 2016-11-05:
You are welcome.

Btw: I know it sometimes is hard to tell in advance if something is a bug report or support question. But if you can please post support question on stack overflow using the "yosys" tag. (I have the theory that for every person who asks a question, there are at least 10 others who have the same question but don't ask. For those 10 other people stackoverflow is imo better suited as archive of old questions and answers than closed issues on github.)

AlexDaniel, 2016-11-05:
@cliffordwolf Actually, given how little information exists for such corner cases, closed github issues show up just fine in google… At least, that's what I've noticed when working with Yosys :)

But sure, I will try to do that next time. That being said, I think this projects lacks a wiki. I will definitely contribute a couple of pages on things that took me forever to figure out by myself… Perhaps consider enabling a wiki on github?

cliffordwolf, 2016-11-06:
> closed github issues show up just fine in google

sure. but the question-answer format of stack overflow is imo much better suited for people looking for the answer later.

> Perhaps consider enabling a wiki on github?

I've made mixed experiences with wikis over the years. They usually need someone who looks after them. Would you be interested in filling this role?

AlexDaniel, 2016-11-06:
OK, I can do that.

cliffordwolf, 2016-11-06:
> OK, I can do that.

I've now enabled the wiki for the yosys repo. Happy wikiing!