YosysHQ/yosys #265: SCC (or Submod) does not work as intended

mithridate, 2016-11-09:
I faced a corner case that SCC pass does not work as intended. I have simplified the test design and also attached the log.

Problem description: when one scc feeds another scc, the pass does not encapsulate them correctly.

[log.txt](https://github.com/cliffordwolf/yosys/files/579958/log.txt)
[verilog.txt](https://github.com/cliffordwolf/yosys/files/579959/verilog.txt)

cliffordwolf, 2016-11-09:
This effectively is an empty module. There is only an undriven output `Z`. None of the stuff in the module contributes to any output signal, assertion, assumption, or kept signal. So the first time `clean` is called its all thrown away. I can't see anything unexpected here. If I e.g. make the wires `b`, `w`, and `t` outputs then `scc` extracts three sub modules, just as one would expect.

Btw: You should always at least run `hierarchy` and `proc` on a design before doing anything else. The recommended procedure is to run `prep` after reading the design, if the various `synth*` commands are not applicable.

mithridate, 2016-11-09:
I guess I am being annoying by my bad skill in representing the problem. My apologies. But for any given design with SCC (for example the one you posted on stackoverflow [http://stackoverflow.com/questions/40443592/how-to-put-a-list-of-cells-into-a-submodule-in-yosys/40444293?noredirect=1#comment68255637_40444293](url) ) or the one that I have attached here), the yosys crashed with the following sequence of commands:

yosys> read_verilog test_2.v
yosys>proc; opt; fsm; opt; memory; opt
yosys> prep
yosys> hierarchy -auto-top
yosys> scc -all_cell_types -set_attr submod scc{}
yosys> submod
yosys> hierarchy -auto-top
yosys> flatten
yosys> scc -all_cell_types -set_attr submod scc{}
yosys> submod
yosys> hierarchy -auto-top
yosys> flatten

11. Executing FLATTEN pass (flatten design).
Mapping test.test_scc2 using test_scc2.
ERROR: Assert `count_id(wire->name) == 0' failed in kernel/rtlil.cc:1313.
mehrdad:dbg_scc$

mithridate, 2016-11-19:
I have tried the following experiment and I found that after the following commands
```
yosys> scc -all_cell_types -set_attr submod scc{}
yosys> submod
```
connections of the newly submoddules in top module is not stored as wire in RTLIL (theoretically they are wires in the parent module).

The experiment is as follows:
1. for a circuit with SCC for example the following, find the scc and encapsulate it to submods.
```
module test(input A, B, output X, Y);
assign X = (A & B) ^ X, Y = A | (B ^ Y);
endmodule
```

2. apply the following function to the design: I wrote a pass that simply writes the wires of the top-module:
``` 
 topModule = design->top_module();
  for (auto wire : topModule->wires()) {
    if (wire->port_input)
      std::cout << "Input port:" << wire->name.c_str() << std::endl;
    else if (wire->port_output)
      std::cout << "Output port:" << wire->name.c_str() << std::endl;
    else
      std::cout << "Just wire:" << wire->name.c_str() << std::endl;
  }
```
3. you will see the following
```
yosys> write_wire_info
Just wire:$and$test_1.v:2$1_Y
Input port:\A
Input port:\B
Output port:\X
Output port:\Y

```
NOTE: the wire connection between SCC_0 and "and" gate (in figure you provided in http://stackoverflow.com/questions/40443592/how-to-put-a-list-of-cells-into-a-submodule-in-yosys) is missing.

4. Now write the same design and read the wirtten file back and apply same functon (after prep; and proc;)
you will see:

```
yosys> write_wire_info
Just wire:$and$test_1_scc.v:9$1_Y
Output port:\Y
Output port:\X
Input port:\B
Input port:\A
**_Just wire:\_0__**

```

you can see a wire (\_0__) that was missing in RTLIL after submod.

I hope it helps.

povik, 2023-09-29:
As far as I can tell the issues described here do not reproduce on current Yosys. I will close the ticket but feel free to raise objections.

closed 2023-09-29 by povik