YosysHQ/yosys #266: Question: Successful macOS builds and brew tap for yosys

secworks, 2016-11-10:
(A bit unsure of Github really is the best place for this question.)

I tried to clone and build HEAD of yosys in macOS (10.12.1). Unfortunately it failed. (I'll post errors and more specifics on environment.). But I saw someone else posting about macOS and thought I should ask before diving in.

Is somebody successfully building yosys on recent macOS?

Has somebody implemented a working tap for homebrew? I searched but but didn't find one. There is none in main brew and none in pull requests for homebrew. Having a working tap would be nice and I'm willing to do so if it doesn't exist yet.

padf, 2016-11-13:
Well, not sure also if this is the right place but I have just compiled yosys 0.7  on MacOs ( but 10.11.6 - ElCapitan). 

Not running homebrew but macports.

Had to make a special port file for iverilog as iverilog is without maintainer and is on version 0.9.7 on macports. 
Had to install Mercurial (with macports). 

Then, I followed the instructions:
```
make config-clang 
make 
make test 
``` 

On the config phase make complained about the -r option on sed (not portable). 
 
Make complained many times about brew not being installed, but compiled fine and all the tests have run ok. 

Here are the warnings found on the compilation phase.  I think only the first ones may  merit some attention. 

```

[ 15%] Building frontends/ilang/ilang_lexer.o
frontends/ilang/ilang_lexer.cc:1462:23: warning: comparison of integers of different signs: 'yy_size_t' (aka 'unsigned long') and
      'int' [-Wsign-compare]
                if ( number_to_move == YY_MORE_ADJ )
                     ~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~
1 warning generated.                      

[ 18%] Building frontends/verilog/verilog_lexer.o
frontends/verilog/verilog_lexer.cc:2427:23: warning: comparison of integers of different signs: 'yy_size_t' (aka 'unsigned long')
      and 'int' [-Wsign-compare]
                if ( number_to_move == YY_MORE_ADJ )
                     ~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~
1 warning generated.       
   
[ 95%] ABC: `` Compiling: /src/map/scl/sclLiberty.c
src/map/scl/sclLiberty.c:586:10: warning: logical not is only applied to the left hand side of this comparison
      [-Wlogical-not-parentheses]
    if ( !Scl_LibertyBuildItem( p, &pPos, p->pContents + p->nContents ) == 0 )
         ^                                                              ~~
src/map/scl/sclLiberty.c:586:10: note: add parentheses after the '!' to evaluate the comparison first
    if ( !Scl_LibertyBuildItem( p, &pPos, p->pContents + p->nContents ) == 0 )
         ^
          (                                                                 )
src/map/scl/sclLiberty.c:586:10: note: add parentheses around left hand side expression to silence this warning
    if ( !Scl_LibertyBuildItem( p, &pPos, p->pContents + p->nContents ) == 0 )
         ^
         (                                                             )
1 warning generated. 
 
                     

[ 95%] ABC: `` Compiling: /src/misc/zlib/inflate.c
src/misc/zlib/inflate.c:1448:61: warning: shifting a negative signed value is undefined [-Wshift-negative-value]
    if (strm == Z_NULL || strm->state == Z_NULL) return -1L << 16;
                                                        ~~~ ^
1 warning generated.

                     
                     
[ 95%] ABC: `` Compiling: /src/proof/acec/acecPolyn.c
src/proof/acec/acecPolyn.c:432:17: warning: unused variable 'vLit2Mono' [-Wunused-variable]
    Vec_Wec_t * vLit2Mono = Vec_WecStart( Gia_ManObjNum(pGia) * 2 );
                ^
1 warning generated. 

```

secworks, 2016-11-14:
Thanks for detailed report. I will try and reproduce on Sierra with Homebrew.

cliffordwolf, 2017-01-04:
Has this issue been resolved?

I don't have a Mac myself, so I cannot help. But at 33C3 we had a few workshops and people with Macs where attending all the time and where able to build Yosys on their systems. So I know it *should* work.

cliffordwolf, 2017-01-13:
> Has this issue been resolved?

closing the issue after >1 week, assuming it has been resolved.

closed 2017-01-13 by cliffordwolf