YosysHQ/yosys #267: FF maping

Gl237man, 2016-11-11:
Sometimes triggers have unrelaten pins in liberty file. During dfflibmap YOSYS searches for exact match with internal primitives to detect FFs. It would be useful to add detection of unused pins in cells from Liberty so that cells with them will be correctly detected as FFs and used during mapping process.

cliffordwolf, 2017-01-04:
Extra outputs on FF cells are already ignored and the cell is used with the extra output left unconnected. When a cell has additional inputs however it is difficult to "ignore" those inputs because we need to drive it with something and we don't know what to drive it with.

Do you have an example of a liberty cell declaration for an FF with additional inputs where you think it would be useful to still use this cell? Can you post the liberty code for this cell and explain what you'd expect the tool to do to figure out how to drive the additional input?

cliffordwolf, 2017-01-13:
Inactive for >1 week. Closing the issue. I will re-open it if and when the requested information is provided.

closed 2017-01-13 by cliffordwolf