YosysHQ/yosys #269: a < 0 can be optimized if a is signed

AlexDaniel, 2016-11-14:
Here is the shortest module to reproduce the issue:
```verilog
module isneg (input signed [7:0] a, output logic o);
   assign o = a < 0;
   //assign o = a[7];
endmodule
```

Given that signed numbers are in two's complement, you can use MSB to see if the number is negative or not (``assign o = a[7]``). Synthesis result in this case should be obvious.

However, for ``a < 0`` yosys produces this:
[image]
(complete command to reproduce exactly the same graph is ``yosys -p 'synth_xilinx; show -colors 67 -format svg -prefix isneg' isneg.sv``)

Sure, I can always optimize it manually, but I find it less readable this way.

cliffordwolf, 2016-11-27:
I'll add a proper peephole optimizer with syntax for a rules files and a code generator that takes rules files and generates the optimizer pass, in addition to or as a replacement for opt_expr. I'll make sure to add this optimization as one of the first rules when this feature becomes available.

I'll update this issue when there is any progress. I hope I will get to it this year, but there is not much year left..

cliffordwolf, 2017-01-31:
This optimization has been implemented by #292, so I'm now closing this issue.
(The generic poophole optimizer stays on my todo list. Not sure when I will get to it though.)

closed 2017-01-31 by cliffordwolf

AlexDaniel, 2017-01-31:
Great! Thanks.