YosysHQ/yosys #270: Problem with compiling under windows10, msys2, gcc 6.2, clang 3.8.

Cinek75, 2016-11-14:
Hello.
Under Windows 10 environment I use msys2 and gcc 6.2, 3.8 clang etc.
After issuing the command $ make config-clang or $ make gcc-config file is created makefile.conf.
Compilation of the program under Windows 10 fails.

$ make
[Makefile.conf] CONFIG: = clang
could not read file "/ proc / 16056 / fd / 63": no such file or directory
[0%] Building kernel / version_UNKNOWN.cc
could not read file "/ proc / 11072 / fd / 63": no such file or directory
[0%] Building kernel / version_UNKNOWN.o
could not read file "/ proc / 10712 / fd / 63": no such file or directory
[1%] Building kernel / driver.o
In file included from kernel / driver.cc 20:
In file included from. \ Kernel / yosys.h: 142:
. \ Kernel / hashlib.h: 150: 10: error: cast from pointer to smaller type 'unsigned long' Loses information
                 return (unsigned long) and;
                        ^ ~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile: 367: kernel / driver.o] Error 1
Greetings.

cliffordwolf, 2016-11-18:
Interesting that this is an error and not a warning.. Does replacing `(unsigned long)` with `(unsigned long long)` help? (I'm not sure about the definition windows uses for the width of those types.) The loss of information does not matter in this case: This is a hash function, using only the lower 32 bits should be fine.

cliffordwolf, 2017-01-04:
OP did not answer my question for nearly 2 months. Closing the issue.

closed 2017-01-04 by cliffordwolf