YosysHQ/yosys #273: Design size increases after flattening

AlexDaniel, 2016-11-22:
So my understanding after [this question](https://stackoverflow.com/questions/40668714/) is that the design size should not increase dramatically after flattening.

> Small changes to the design, even when they seem irrelevant, can cause small random fluctuations in design size

That is true. In fact, this problem easily goes away if you tweak the description just a little bit.

I've noticed this issue many times, and here is the example with the most dramatic difference I've seen so far.

Please don't mind what is happening in the description, I know it can be improved significantly. This is just something to replicate the issue. I didn't really succeed in golfing it, hope you'll have better luck.

```
Yosys 0.7+3 (git sha1 2874914, gcc 6.2.0-9 -fPIC -Os)
```

```bash
yosys -p 'synth_xilinx -top gcd' gcd.sv                # Number of cells: 144
yosys -p 'synth_xilinx -flatten -top gcd' gcd.sv       # Number of cells: 191
```

```verilog
module gcd (input clk, input reset,
            input signed [N-1:0] a,
            input signed [N-1:0] b,
            input input_available,
            input result_taken,
            output logic idle,
            output logic result_available,
            output logic [N-1:0] out);
   parameter N = 8;

   wire       a_en;
   wire       b_en;
   wire [1:0] a_sel;
   wire       b_sel;
   wire       dp_zero;
   wire       dp_lt;

   gcd_datapath #(.N(N)) gcd_datapath
     (
      .clk(clk),
      .reset(reset),
      .a(a),
      .b(b),
      .a_en(a_en),
      .b_en(b_en),
      .a_sel(a_sel),
      .b_sel(b_sel),
      .dp_zero(dp_zero),
      .dp_lt(dp_lt),
      .result_data(out)
      );

   gcd_state gcd_state
     (
      .clk(clk),
      .reset(reset),
      .input_available(input_available),
      .result_taken(result_taken),
      .idle(idle),
      .result_available(result_available),
      .dp_zero(dp_zero),
      .dp_lt(dp_lt),
      .a_en(a_en),
      .b_en(b_en),
      .a_sel(a_sel),
      .b_sel(b_sel)
      );
endmodule

module gcd_state (input clk, input reset,
                  input              input_available,
                  input              result_taken,
                  output logic       idle,
                  output logic       result_available,
                  input              dp_zero,
                  input              dp_lt,
                  output logic       a_en,
                  output logic       b_en,
                  output logic [1:0] a_sel,
                  output logic       b_sel
                  );
   localparam
     STATE_WAIT = 2'd0,
     STATE_CALC = 2'd1,
     STATE_DONE = 2'd2;

   (* fsm_encoding = "auto" *)
   logic [1:0] state;
   logic [1:0] next_state;

   always @(posedge clk)
     if (reset)
       state = STATE_WAIT;
     else
       state = next_state;

   always @* begin
      next_state = state;
      result_available = 0;
      idle = 0;
      a_en = 0; a_sel = 0;
      b_en = 0; b_sel = 0;

      (* parallel_case *) (* full_case *)
      case (state)
        STATE_WAIT: begin
           a_en = 1; a_sel = 2'b00;
           b_en = 1; b_sel = 2'b00;

           idle = 1;
           if (input_available)
             next_state = STATE_CALC;
        end
        STATE_CALC: begin
           if (dp_lt) begin
              a_en = 1; a_sel = 2'b01;
              b_en = 1; b_sel = 2'b01;
           end else if (!dp_zero) begin
              a_en = 1; a_sel = 2'b10;
           end
           if (dp_zero)
              next_state = STATE_DONE;
        end
        STATE_DONE: begin
           result_available = 1;
           if (result_taken)
              next_state = STATE_WAIT;
        end
       endcase
   end
endmodule

module gcd_datapath (input clk, input reset,
                     input signed [N-1:0] a,
                     input signed [N-1:0] b,
                     input        a_en,
                     input        b_en,
                     input  [1:0] a_sel,
                     input        b_sel,
                     output logic dp_zero,
                     output logic dp_lt,
                     output logic [N-1:0] result_data);
   parameter N = 8;

   logic signed [N:0] sub;

   logic signed [N:0] a_out;
   logic signed [N:0] b_out;

   always @(posedge clk) begin
      // A mux
      if (a_en) begin
         if (a_sel[1])
           a_out <= sub;
         else
           a_out <= a_sel[0] ? b_out : (a[N-1] ? -a : a);
      end
      // B mux
      if (b_en)
        b_out <= b_sel ? a_out : (b[N-1] ? -b : b);
   end

   assign dp_lt   = a_out < b_out;
   assign dp_zero = b_out == 0;
   assign sub     = a_out - b_out;

   assign result_data = a_out[N-1:0];
endmodule
```

cliffordwolf, 2019-03-02:
This test case is amazingly stable considering it was originally reported 2.5 years ago.  ;)

In general, cell count is not a very good indicator of area, you'd need to ignore those MUXF7/MUXF8 cells in your count, figure out which FFs can share a slice with a LUT, and check with pairs of LUTs can actually be implemented in just one LUT5_2.

But it turns out, even if I do all that we still get 25 slices (flat) vs 18 slices (hierarchical): http://svn.clifford.at/handicraft/2019/vivadogcdsize

We are right now working on whitebox support in logic mapping. Maybe this will improve this fluke.

AlexDaniel, 2019-03-03:
@cliffordwolf thank you for your work!

lethalbit, 2022-08-18:
As of yosys version 0.20 the cell count in general has been decreased on that design.

`126` for `yosys -p 'synth_xilinx -top gcd' gcd.sv`
`141` for `yosys -p 'synth_xilinx -flatten -top gcd' gcd.sv`

However passing `-abc9` with the `synth_xilinx` improves the cell count.

`123` for `yosys -p 'synth_xilinx -abc9 -flatten -top gcd' gcd.sv` and `yosys -p 'synth_xilinx -abc9 -top gcd' gcd.sv`

and `111` for `yosys -p 'synth_xilinx -abc9 -dff -flatten -top gcd' gcd.sv`

So the cell counts in general have been improved quite a lot, and the flatended design can be minimized even further with the addition of `-abc9 -dff`.

Hope this helps.

closed 2022-08-18 by lethalbit