YosysHQ/yosys #280: "0" in connection section of JSON

drom, 2016-11-27:
When converting BLIF into JSON (pre or post PNR) the `cell.connections` object has following structure:

```json
          "connections": {
            "O": [ 67 ],
            "I3": [ 66 ],
            "I2": [ 43 ],
            "I1": [ "0" ],
            "I0": [ "0" ]
          }
```
post-pack
```json
          "connections": {
            "COUT": [ ],
            "O": [ 85 ],
            "LO": [ ],
            "SR": [ ],
            "CEN": [ ],
            "CLK": [ ],
            "CIN": [ ],
            "I3": [ "0" ],
            "I2": [ "0" ],
            "I1": [ 91 ],
            "I0": [ 92 ]
          }
```
All values except 0 are numbers. Is any specific reason why "0" is a string?
Also I see that each connection is an array of size 0 or 1. Do you need array? Can we have more then 1 values?
JSON supports `null` as the value. Does is make sense to use plain numbers and `null` for the empty value?

cliffordwolf, 2016-11-27:
> All values except 0 are numbers. Is any specific reason why "0" is a string?

"0", "1", and "x" are constants. Nets start with 2 in order to avoid confusion with JSON parsers that automatically convert between string and numbers. (I don't know of any, but better safe than sorry.)

> Also I see that each connection is an array of size 0 or 1. Do you need array?

Yes. In the iCE40 case all ports are single-bit wide. But in other cases the ports may be wider.

closed 2016-11-27 by cliffordwolf

drom, 2016-11-28:
It does make sense. Thank you for explanation.