YosysHQ/yosys #285: Probem cross-compiling ABC (From Yosys)

udif, 2016-12-26:
I'm trying to cross compile Yosis for Windows using docker and the mxe target;

```
FROM debian:jessie
MAINTAINER Udi Finkelstein <github@udifink.om>
RUN \
  echo "deb http://pkg.mxe.cc/repos/apt/debian jessie main" > /etc/apt/sources.list.d/mxeapt.list && \
  apt-key adv --keyserver keyserver.ubuntu.com --recv-keys D43A795B73B16ABE9643FE1AFD8FFF16DB45C6AB && \
  apt-get update -y && \
  apt-get install -y \
  autoconf \
  bison \
  flex \
  g++ \
  gawk \
  gettext \
  git \
  libffi-dev \
  make \
  mercurial \
  mxe-i686-w64-mingw32.static-gcc \
  pkg-config \
  python3 \
   tcl-dev \
  xdot

RUN git clone https://github.com/cliffordwolf/yosys.git && cd yosys && make config-mxe && \
  ln -s /usr/lib/mxe /usr/local/src/mxe && \
  make mxebin

RUN git clone https://github.com/cliffordwolf/icestorm.git icestorm && cd icestorm && make -j$(nproc) mxebin

RUN git clone https://github.com/cseed/arachne-pnr.git && cd arachne-pnr && make mxebin
```

Compilation progresses fine, until I get the following error while compiling ABC:

```
src/proof/acec/acecSt.o: In function `Z17Acec_StatsCollectP10Gia_Man_t_i':
/yosys/abc/src/proof/acec/acecSt.c:409: undefined reference to `Dar_LibReturnClass(unsigned int)'
collect2: error: ld returned 1 exit status
make[1]: *** [abc-a4872e22c646.exe] Error 1
```

After some investigation it seems that the cross compile configs always use '-x c++' causing all files to be parsed as C++ (despite their '.c' suffix).
The file defining Dar_LibReturnClass() (sorry, forgot the path for this file) is compiled as C, while src/proof/acec/acecSt.c is compiled as C++, assuming Dar_LibReturnClass() is a C++ func and calls it with a mangled filename.
The non-cross-compile versions successfully compile ABC as C code.
This also happened when I tried to use the Ubuntu built-in mingw32 cross compile packages (I created soft links from /usr/bin/mingw... to the hardcoded path in the Yosys Makefile).

I tracked the source of the '-x c++' flag and it dates back to the original commit of the cross-compile code in the Makefile.
Is there any special reason you compiled ABC as C++?

cliffordwolf, 2016-12-26:
I'll see that I get a build fix for that into ABC upstream.

> Is there any special reason you compiled ABC as C++?

Yes. Copy & paste from an explanation I gave somewhere else:

```
the reason I'm building ABC with "-x c++" when cross-compiling for
windows is that in src/base/cmd/cmd.c there is:

        1148 #if defined(WIN32) && !defined(__cplusplus)
        1149 #include <direct.h>
        1150
        1151 // these structures are defined in <io.h> but are for some reason invisible
        1152 typedef unsigned long _fsize_t; // Could be 64 bits for Win32
        1153
        1154 struct _finddata_t {
        1155     unsigned    attrib;
        1156     time_t      time_create;    // -1 for FAT file systems
        1157     time_t      time_access;    // -1 for FAT file systems
        1158     time_t      time_write;
        1159     _fsize_t    size;
        1160     char        name[260];
        1161 };
             ...

however, the io.h from MXE defines those things just fine in C mode,
resulting in compiler errors for redefined data types:

        src/base/cmd/cmd.c:1154:8: error: redefinition of 'struct _finddata_t'
         struct _finddata_t {
                ^
        In file included from
        /usr/local/src/mxe/usr/i686-pc-mingw32/include/direct.h:22:0,
                         from src/base/cmd/cmd.c:1149:
        /usr/local/src/mxe/usr/i686-pc-mingw32/include/io.h:63:8: note: originally
        defined here
         struct _finddata_t
                ^
        ...

By using "-x c++" I force __cplusplus to be defined and am thus avoiding
that errors.
```

cliffordwolf, 2016-12-26:
The following patch for ABC fixes the issue. I already sent it upstream to the ABC maintainer.

```
diff -r f591c081d5e7 src/proof/acec/acecSt.c
--- a/src/proof/acec/acecSt.c	Sun Dec 25 15:58:54 2016 +0700
+++ b/src/proof/acec/acecSt.c	Mon Dec 26 18:12:16 2016 +0100
@@ -21,6 +21,8 @@
 #include "acecInt.h"
 #include "misc/vec/vecWec.h"
 #include "misc/extra/extra.h"
+#include "aig/aig/aig.h"
+#include "opt/dar/dar.h"
 
 ABC_NAMESPACE_IMPL_START
 
```

udif, 2016-12-27:
Thanks for the quick reply!
I understand the reason for using  `-x c++`, but somehow this solution looks like a workaround instead of a real solution to this problem, so instead of patching ABC so it can compile as C++ again, we could patch the real problem preventing this C program from compiling as C.

The explanation in `aceSt.c` is pretty vague. It is not clear WHICH `<io.h>` is making the problems for ABC, and why exactly are the structure really invisible there and not just "for some reason".
Looking at my copy of Visual Studio, I see:
```
#ifdef _USE_32BIT_TIME_T
    #define _finddata_t     _finddata32_t
    #define _finddatai64_t  _finddata32i64_t
#else
    #define _finddata_t     _finddata64i32_t
    #define _finddatai64_t  __finddata64_t
#endif

struct _finddata32_t
{
    unsigned    attrib;
    __time32_t  time_create;    // -1 for FAT file systems
    __time32_t  time_access;    // -1 for FAT file systems
    __time32_t  time_write;
    _fsize_t    size;
    char        name[260];
};

struct _finddata32i64_t
{
    unsigned    attrib;
    __time32_t  time_create;    // -1 for FAT file systems
    __time32_t  time_access;    // -1 for FAT file systems
    __time32_t  time_write;
    __int64     size;
    char        name[260];
};

struct _finddata64i32_t
{
    unsigned    attrib;
    __time64_t  time_create;    // -1 for FAT file systems
    __time64_t  time_access;    // -1 for FAT file systems
    __time64_t  time_write;
    _fsize_t    size;
    char        name[260];
};

struct __finddata64_t
{
    unsigned    attrib;
    __time64_t  time_create;    // -1 for FAT file systems
    __time64_t  time_access;    // -1 for FAT file systems
    __time64_t  time_write;
    __int64     size;
    char        name[260];
};
```
This is ofcourse only part of the file, but the code above was not surrounded by ANY `#i`f/`#ifdef`'s.
I really see no reason for the private `_finddata_t` definition.
The only problem with the ABC code from cmd.c is that it is including `<direct.h>` but expecting to find structs from `<io.h>`
I went over Visual Studio 2015's <direct.h> amd while it does include other files, <io.h> is not one of them, which is the reason why `_finddata_t` was not defined. mingw64 OTOH, includes `<io.h>` from `<dirent.h>`. I think that all that's needed for the fix is to explicitly add `#include <io.h>` to `cmd.c`.

Actually, it seems that both cmd.c and aceSt.c should be fixed, since regardless of any other workaround, `cmd.c` should not call `Dar_LibReturnClass()` without a proper function prototype.

Running `git blame` on `acecSt.c` reveals that it is a relatively new file, from Oct 22, 2016. This probably means that cross-compilation was broken since that check-in.

In any case, thanks for making it work, regardless of the way!

udif, 2016-12-28:
It seems that the ABC pull request has been applied manually.
https://bitbucket.org/alanmi/abc/commits/aea778a02d014f108ee2684526709a7da9ad873e

cliffordwolf, 2016-12-29:
updated ABC in yosys.

closed 2016-12-29 by cliffordwolf