YosysHQ/yosys #287: Locally defined arrays not recognized

secworks, 2017-01-03:
I'm trying to implement my ChaCha core (https://github.com/secworks/chacha) with Yosys and bumped into one issue. It seems two dimensional arrays defined in local scope is not recognized.

In chacha_core.v, data_out_logic process (row 494-517 in https://github.com/secworks/chacha/blob/master/src/rtl/chacha_core.v) I define local arrays:

      reg [31 : 0] msb_block_state [0 : 15];
      reg [31 : 0] lsb_block_state [0 : 15];
      reg [511 : 0] block_state;

running read_verilog I get:
Warning: Identifier `\data_out_logic.msb_block_state' is implicitly declared at chacha_core.v:507.
Warning: Identifier `\data_out_logic.lsb_block_state' is implicitly declared at chacha_core.v:508.

Moving the array definitions to top level scope in the file make yosys being able to find the definitions. So it seems the one dimensional array is ok to define locally, but two dimensional arrays are not.

cliffordwolf, 2017-01-03:
Simple self-contained test case:

```
module test(input [6:0] D, input [1:0] S, output reg [1:0] Y);
  always @* begin
    reg [3:0] data [0:3];
    data[0] = D[3:0];
    data[1] = D[4:1];
    data[2] = D[5:2];
    data[3] = D[6:3];
    Y = data[S];
  end
endmodule
```

secworks, 2017-01-03:
Tested it with yosys 0.5+:

yosys> read_verilog test.v
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `test.v' to AST representation.
Generating RTLIL representation for module `\test'.
Warning: Identifier `\data' is implicitly declared at test.v:4.
Warning: Range select out of bounds on signal `\data' at test.v:4: Setting result bit to undef.
Warning: Range select out of bounds on signal `\data' at test.v:5: Setting result bit to undef.
Warning: Range select out of bounds on signal `\data' at test.v:6: Setting result bit to undef.
Warning: Range select out of bounds on signal `\data' at test.v:7: Setting result bit to undef.
ERROR: Abort in frontends/ast/genrtlil.cc:528.

Seems to be a good test.

cliffordwolf, 2017-01-04:
Commit 81a9ee2 adds support for local memory declarations. Your original HDL code should be working fine now (Yosys 0.7+62).

However, my "simple test case" has local declarations in an *unnamed* block, which is a SystemVerilog feature that is not supported. Commit 81a9ee2 also adds a proper error message for this.

closed 2017-01-04 by cliffordwolf