YosysHQ/yosys #288: yosys-smtbmc relies on POSIX select and poll

cr1901, 2017-01-03:
I'm trying to run `yosys-smtbmc` on Windows, and have run into a problem in stmio.py where check_sat waits for I/O to complete by `select`ing `stdout`. On Windows, `select` is only supported on TCP/IP sockets, and the analogous I/O facility (IOCP) isn't supported in Python's standard library alone.

Furthermore, even if I fix this issue, I would run into problems when calling poll during read(), which is not supported on Windows at all.

I'm willing to fix this myself, but as usual, the problem isn't that simple to fix, and it's been a while since I've had to deal with IPC. Just wanted to make an issue to notify.

cr1901, 2017-01-04:
Workaround for now:

1. Open `$YOSYS_ROOT/share/python3/smtio.py` and modify https://github.com/cliffordwolf/yosys/blob/master/backends/smt2/smtio.py#L636 so, `self.timeinfo` is false (Is this a bug?).

2. Pass `--no-progress` to all invocations to `yosys-smtbmc` for now. Not ideal, but at least it works.

cliffordwolf, 2017-01-04:
Fixed `--no-progress` handling in commit ed812ea.

> or just run z3 directly if you need progress reports

I don't understand. How would you "run z3 directly" and in which way would that help restoring the functionality that `--no-progress` disables?

> Not ideal, but at least it works.

So calling `self.p.poll()` to figure out if the solver has terminated in `read()` is not a problem after all? In your first post you said it is "not supported on Windows at all".

cliffordwolf, 2017-01-04:
> so if getting feedback from z3 was important

huh?

There seems to be a massive misunderstand re what not using `--no-progress` does. Obviously there is no extra information passed from the SMT solver to the user. All it does it displaying and updating a little timer display that informs the user for how long we are already waiting for new results from the solver.

As you can *very easily* see from the `while select(...):` loop, all it does is waiting for new data from the solver becoming available and display a timer. With `not self.timeinfo` this will simply be skipped and the code calls `read()` directly, which will just block silently until data becomes available.

I.e. by default the code produces the last the last line in this "screenshot":

```
##     12   0:00:12  Checking asserts in step 10..
##     18   0:00:18  Checking asserts in step 11..
##     23   0:00:23  Checking asserts in step 12..
 (3 seconds) \
```

With `--no-progress` it will just sit there saying `##     23   0:00:23  Checking asserts in step 12..` and you wouldn't know how old that message is. That's all.

*Nothing* would be gained in this regard if you'd call z3 with the generated smt2 code directly.

cr1901, 2017-01-04:
I should file bug reports when I'm tired/just woke up. Yes, you're right; that loop doesn't interact with the subprocess in any way 'cept for the select statement.

I managed to convinced myself that flushing sys.stderr() flushed z3's stderr until I realized that:
1. Popen redirected z3's stderr to stdout.
2. p.stderr.flush() would be the correct statement I think.

In any case, prob safe to just disable on Windows "for technical reasons".

cliffordwolf, 2017-01-04:
> In any case, prob safe to just disable displaying the timer on Windows "for technical reasons".

done in commit b9ad91b.

closed 2017-01-04 by cliffordwolf