YosysHQ/yosys #294: Segmentation fault when performing read_verilog on large file

mhoeher, 2017-01-17:
When trying to perform read_verilog on a file that contains approx. 115000 lines, yosys crashes with an unspecific segmentation fault. As this file only contains very primitive verilog constructs (no functions, no systemverilog etc.), I assume the crash happens mainly because of the file size. The specialty of the file is the fact that it not only contains one module but rather many module instances and definitions. As it is part of delivered IP, I cannot easily modify it or publish it here. The file contains actual RTL, no netlist.

Maybe it is possible for you to recreate this anyway, let me know if you need further input.

cliffordwolf, 2017-01-17:
Here is a large Verilog file that I created using [rocket-chip](https://github.com/ucb-bar/rocket-chip): [test_v.zip](https://github.com/cliffordwolf/yosys/files/710689/test_v.zip)

It is 11 MB in size, 357784 lines, 323 modules (288 used by ExampleRocketTop). On my machine Yosys completes reading this in less than 6 seconds, no problem. I'm gonna need more information that simply "it's a large file".

I understand if you cannot share the design you are having problems with. But without an example to reproduce the problem I'm not going to be able to do anything about it.

Maybe you can debug this on your own using the file you have and use your findings to generate an artificial test case that you are allowed to share.

> Maybe it is possible for you to recreate this anyway, let me know if you need further input.

Any reason in particular why you did not try this yourself before posting?

I'm closing this issue for now. I'll reopen it if and when someone posts a test case.

markusdd, 2017-01-17:
Hi, I just dropped you an e-mail containing a stripped-down version of the file with mangled names, which I was allowed to share with you personally.

cliffordwolf, 2017-01-17:
Thanks for the test case per email!

This test case contains single expressions that are >40 kB in size and in the order of 1k parenthesis levels deep. The `read_verilog` command runs out of stack space when trying to traverse the AST for this huge expressions. Simply running Yosys with a larger stack (e.g. `ulimit -s 32768`) seems to fix the issue.

closed 2017-01-17 by cliffordwolf

markusdd, 2017-01-17:
I can confirm that this works.

Thanks very much for the extraordinary quick help!