YosysHQ/yosys #295: for loop bad error

OrDavidi, 2017-01-17:
compiling the following module gets this error : 
2nd expression of generate for-loop at test.v:52 is not constant!

module temp

#(parameter size=4, logsize=2)
(
  input  [logsize-1:0]    p_enc,  
  output  wire [size-1:0]    y
  );

  reg     [logsize-1:0] x;
  reg     [2**logsize -1:0] y1;
  integer pow2;
  reg     tmp;
  integer i;
  integer j;
  
  always@*
  begin
    x = p_enc;
    pow2 = 1;
    y1[0] = 0;
    
    for(i=0; i<logsize; i=i+1) begin
      for(j=0; j<pow2; j=j+1) begin /* line 52 with the error ( pow2 is constant but changes each i loop */
        tmp = y1[j];
        y1[j] = tmp | x[i];
        y1[j+pow2] = tmp & x[i];      
      end 
      pow2 = pow2+pow2;
    end
  end  
  assign y = y1[size-1:0];

endmodule

cliffordwolf, 2017-01-17:
This is a wontfix. regs (this includes integers) are never a constant for the AST transformations that unroll for loops. Changing this would require a major redesign of the way the verilog front-end is designed. The only regs that are recognized as constants by the AST transformations are the loop variables themselves.

Simply replace `pow2` with something like `(1 << i)` in this simple example.

closed 2017-01-17 by cliffordwolf

OrDavidi, 2017-01-17:
no problem. Thanks!

haggaie, 2018-06-15:
Hi,

I got a similar issue with Vivado HLS autogenerated code. When generating an AXI4-MM interface, it generates the following code:
```verilog
module top_mem_V_m_axi_decoder
#(parameter
    DIN_WIDTH       = 3
)(
    input  wire [DIN_WIDTH-1:0]         din,
    output reg  [2**DIN_WIDTH-1:0]      dout
);
    integer i;
    always @(din) begin
        dout = {2**DIN_WIDTH{1'b0}};
        for (i=0; i < din; i = i + 1)
            dout[i] = 1'b1;
    end
endmodule
```
I would prefer not to change the Verilog code, as it is autogenerated.

cliffordwolf, 2018-06-15:
> I would prefer not to change the Verilog code, as it is autogenerated.

I can understand that, however, in this case this is not even synthesizable code.

Quoting IEEE Std 1364.1-2002 section 7.7.6 Looping statements:

> Loop bounds shall be statically computable for a for loop.

The value of `din` obviously is not statically computable as it is an input, therefore this is not synthesizable code.

haggaie, 2018-06-16:
I'm not familiar with the spec, and I'm not sure what it means to be statically computable in this case, but I can tell you that Vivado 2016.2 is able to synthesize this code. Perhaps it is some non-standard feature between Vivado and Vivado HLS?

jeremyherbert, 2018-06-16:
“Statically computable” in this case means that the value is known at compile time (or synthesis time). As your loop termination condition “i < din” depends on a value which cannot be known at compile time (the wire din), this is not statically computable and therefore is not valid verilog 2002.

haggaie, 2018-06-16:
Okay, I'll try to take it with Xilinx. Thanks.

alaindargelas, 2023-12-18:
This case:

```
module top_mem_V_m_axi_decoder
#(parameter
    DIN_WIDTH       = 3
)(
    input  wire [DIN_WIDTH-1:0]         din,
    output reg  [2**DIN_WIDTH-1:0]      dout
);
    integer i;
    always @(din) begin
        dout = {2**DIN_WIDTH{1'b0}};
        for (i=0; i < din; i = i + 1)
            dout[i] = 1'b1;
    end
endmodule
```

is now supported here in the SV plugin:
https://github.com/chipsalliance/synlig/pull/2125