YosysHQ/yosys #298: typedef support

OrDavidi, 2017-01-19:
Hi,
I encounter some code with the typedef syntax.
Any idea of working around it ? maybe some preprocessor tricks?  or maybe you have a future plan to support it ?

for example : 
typedef reg [127:0] data_t; 

Thanks

cliffordwolf, 2019-06-27:
Continuing discussion from #1110 @towoe:

> If you would be open to suggest a different approach I would be happy to implement it. Should we discuss this in the issue instead?

The biggest issue here I think is being able to parse variable declarations that simply are `<some_identifier> <some_identifier>;` without knowing already in the lexer that the 1st identifier refers to a type. Currently this would result in reduce/reduce conflicts (or at least shift/reduce conflicts) with the rules for parsing instances.

Spending some time on figuring out how to deal with this mess would be a good first step I think.

jjj11x, 2020-01-17:
I think https://github.com/YosysHQ/yosys/pull/1389 mostly fixes this issue, could close it. The typedef'd typenames still have to be enclosed in parentheses when used (even the tests do so), but other than that everything seems to work for me.

zachjs, 2021-03-16:
The outstanding portion of this issue was resolved in #1803.

closed 2021-03-16 by zachjs