YosysHQ/yosys #299: Adder-Substractor inference

drom, 2017-01-21:
Adder-Subtractor inference is one of common transformations done by synthesis tools.

https://en.wikipedia.org/wiki/Adder%E2%80%93subtractor

From this Verilog:
```verilog
assign res = sel
  ? (a - b)
  : (a + b);
```
Full version: https://github.com/drom/icicles/blob/master/src/add_sub.v

Synthesizer can infer something like this:

```verilog
assign sum = {a, sel} + {(sel ? ~b : b), 1'b1};
assign res = sum[W:1];
```
Full version: https://github.com/drom/icicles/blob/master/src/add_sub_opt.v

Second version requires ~2x less LUTs and faster too.

Does such optimizations belongs to Yosys project?

cliffordwolf, 2017-01-21:
The following command will map your `add_sub.v` to 15 `SB_CARRY` and 32 `SB_LUT4` cells:

```
yosys -p 'proc; opt; alumacc; share -aggressive; synth_ice40' add_sub.v
```

To do this out-of-the-box I will need to write a resource sharing pass that is less universal (but faster) than the `share` pass. That new pass would only be looking at cells driving the same mux (tree). This is on my todo list (proposed name: `opt_share`, this is the reason why I renamed the old `opt_share` to `opt_merge` in March 2016).

I'll keep this issue open until I'll find the time to write that new pass.

drom, 2017-01-21:
LGTM

akashlevy, 2025-02-26:
This appears to be resolved by https://github.com/YosysHQ/yosys/pull/1303