YosysHQ/yosys #300: Mem OFFSET is uninitialized.

Kmanfi, 2017-01-25:
Reading .ilang  where memories are $memrd and $memwr, internal memory OFFSET parameter seems to be uninitialized. The uninitialized offset will show as $mem parameter after memory_collect.

cliffordwolf, 2017-01-25:
I cannot reproduce this. Using the following Verilog code as input (test.v):

```
module top(input clk, input [7:0] addr, wdata, output [7:0] rdata);
  reg [7:0] memory [30:45];
  assign rdata = memory[addr];
  always @(posedge clk) memory[addr] <= wdata;
endmodule
```

Write ilang with unpacked memories:

    yosys -p 'proc; opt -fast; write_ilang test.il' test.v

Load ilang, pack and print $mem cell:

    yosys -p 'memory_collect; dump t:$mem' test.il

This is the $mem cell in the output:

```
  cell $mem \memory
    parameter \RD_TRANSPARENT 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_PORTS 1
    parameter \WR_CLK_POLARITY 1'0
    parameter \WR_CLK_ENABLE 1'0
    parameter \WR_PORTS 1
    parameter \INIT 1'x
    parameter \ABITS 6
    parameter \SIZE 16
    parameter \OFFSET 30
    parameter \WIDTH 8
    parameter \MEMID "\\memory"
    connect \RD_EN 1'x
    connect \RD_DATA \rdata
    connect \RD_ADDR \addr [5:0]
    connect \RD_CLK 1'x
    connect \WR_EN 8'11111111
    connect \WR_DATA $memwr$\memory$test.v:4$1_DATA
    connect \WR_ADDR $memwr$\memory$test.v:4$1_ADDR [5:0]
    connect \WR_CLK 1'x
  end
```

As you can see, OFFSET is 30 as expected.

Please post Verilog code and Yosys script that demonstrate the issue you are seeing.

mtuomi, 2017-01-25:
The problem seems to appear when there is no offset used. That's when ilang doesn't store offset 0 explicitly.

Try with this code : 
module top(input clk, input [7:0] addr, wdata, output [7:0] rdata);
  reg [7:0] memory [255:0];
  assign rdata = memory[addr];
  always @(posedge clk) memory[addr] <= wdata;
endmodule

And with this syn script : 
read_verilog top.v
proc; opt -fast; write_ilang test.il

design -reset; read_ilang test.il; dump m:*
design -reset; read_ilang test.il; dump m:*
design -reset; read_ilang test.il; dump m:*
design -reset; read_ilang test.il; dump m:*
design -reset; read_ilang test.il; dump m:*
design -reset; read_ilang test.il; dump m:*

This printed on my machine : 

....
4. Executing ILANG backend.
Output filename: test.il

5. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 65793 \memory

6. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 1879048192 \memory

7. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 1879048192 \memory

8. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 1879048192 \memory

9. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 268437498 \memory

10. Executing ILANG frontend.
Input filename: test.il

  attribute \src "top.v:2"
  memory width 8 size 256 offset 1879048192 \memory

Kmanfi, 2017-01-25:
```
read_verilog simple_sram_byte_en.v
proc
opt
stat
dump
write_ilang foo.ilang
memory_collect 
stat
dump t:$mem
```

 cell $mem \mem
    parameter \RD_TRANSPARENT 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_PORTS 1
    parameter \WR_CLK_POLARITY 4'0000
    parameter \WR_CLK_ENABLE 4'0000
    parameter \WR_PORTS 4
    parameter \INIT 1'x
    parameter \ABITS 4
    parameter \SIZE 16
    parameter \OFFSET 0
    parameter \WIDTH 32
    parameter \MEMID "\\mem"
    connect \RD_EN 1'x
    connect \RD_DATA $0\o_read_data[31:0]
    connect \RD_ADDR \i_address
    connect \RD_CLK 1'x
    connect \WR_EN { $memwr$\mem$simple_sram_byte_en.v:22$8_EN $memwr$\mem$simple_sram_byte_en.v:22$7_EN $memwr$\mem$simple_sram_byte_en.v:22$6_EN $memwr$\mem$simple_sram_byte_en.v:22$5_EN }
    connect \WR_DATA { $memwr$\mem$simple_sram_byte_en.v:22$8_DATA $memwr$\mem$simple_sram_byte_en.v:22$7_DATA $memwr$\mem$simple_sram_byte_en.v:22$6_DATA $memwr$\mem$simple_sram_byte_en.v:22$5_DATA }
    connect \WR_ADDR { $memwr$\mem$simple_sram_byte_en.v:22$8_ADDR $memwr$\mem$simple_sram_byte_en.v:22$7_ADDR $memwr$\mem$simple_sram_byte_en.v:22$6_ADDR $memwr$\mem$simple_sram_byte_en.v:22$5_ADDR }
    connect \WR_CLK 4'xxxx
  end
```

design -reset
read_ilang foo.ilang 
stat
memory_collect 
dump t:$mem
```
  cell $mem \mem
    parameter \RD_TRANSPARENT 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_PORTS 1
    parameter \WR_CLK_POLARITY 4'0000
    parameter \WR_CLK_ENABLE 4'0000
    parameter \WR_PORTS 4
    parameter \INIT 1'x
    parameter \ABITS 4
    parameter \SIZE 16
    parameter \OFFSET 65792
    parameter \WIDTH 32
    parameter \MEMID "\\mem"
    connect \RD_EN 1'x
    connect \RD_DATA $0\o_read_data[31:0]
    connect \RD_ADDR \i_address
    connect \RD_CLK 1'x
    connect \WR_EN { $memwr$\mem$simple_sram_byte_en.v:22$8_EN $memwr$\mem$simple_sram_byte_en.v:22$7_EN $memwr$\mem$simple_sram_byte_en.v:22$6_EN $memwr$\mem$simple_sram_byte_en.v:22$5_EN }
    connect \WR_DATA { $memwr$\mem$simple_sram_byte_en.v:22$8_DATA $memwr$\mem$simple_sram_byte_en.v:22$7_DATA $memwr$\mem$simple_sram_byte_en.v:22$6_DATA $memwr$\mem$simple_sram_byte_en.v:22$5_DATA }
    connect \WR_ADDR { $memwr$\mem$simple_sram_byte_en.v:22$8_ADDR $memwr$\mem$simple_sram_byte_en.v:22$7_ADDR $memwr$\mem$simple_sram_byte_en.v:22$6_ADDR $memwr$\mem$simple_sram_byte_en.v:22$5_ADDR }
    connect \WR_CLK 4'xxxx
  end

cliffordwolf, 2017-01-25:
> The problem seems to appear when there is no offset used. That's when ilang doesn't store offset 0 explicitly.

Ah! Thanks! Now I understand. Fixed in b54972c.

closed 2017-01-25 by cliffordwolf