YosysHQ/yosys #301: `use of undeclared identifier` error on OSX Sierra

beriberikix, 2017-01-29:
When I try running `make` I get a `use of undeclared identifier` error copied below. I've tried running make with both clang and gcc enabled in the makefile. My machine:
* OSX Sierra 10.12.2
* clang -v Apple LLVM version 8.0.0 (clang-800.0.42.1)

Any suggestions?

```
jberi-macbookpro:yosys 🤖  $ make
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
[  0%] Building kernel/version_e54c355b.cc
[  0%] Building kernel/version_e54c355b.o
[  1%] Building kernel/driver.o
kernel/driver.cc:510:4: error: use of undeclared identifier 'append_history'
                        append_history(where_history() - history_offset, history_f...
                        ^
kernel/driver.cc:516:27: error: use of undeclared identifier 'history_list'
        HIST_ENTRY **hist_list = history_list();
                                 ^
2 errors generated.
make: *** [kernel/driver.o] Error 1
```

cliffordwolf, 2017-01-29:
You can always set `ENABLE_READLINE := 0` in your `Makefile.conf` to disable readline history altogether. Maybe [someone else will give a more satisfying answer](https://twitter.com/oe1cxw/status/825659912965472256). I know that people usually can build Yosys with readline support on OSX without problems. I have no idea why `<readline/history.h>` on your system does not provide `append_history()` and `history_list()`.

blanham, 2017-01-29:
Macs use NetBSD's libedit instead of GNU readline, and libedit doesn't have `append_history()` or `history_list()`. Easiest fix is to install GNU readline using Homebrew, and then add Homebrew's lib/ directory to the build.

closed 2017-01-29 by cliffordwolf

beriberikix, 2017-01-29:
It always pops in your head during the middle of the night...the issue is I didn't install homebrew in `/usr/local` to avoid the `sudo` issue. It has worked fine until now but it looks like the makefile hardcodes some of the paths. I mucked around in the makefile to try to fix it but I reinstalled brew back at `/usr/local` and make && make install work fine.

If anyone has better make/shell fu they could probably figure out how to better generalize the makefile.