YosysHQ/yosys #302: Equivalence Check Issue

bobnewgard, 2017-01-30:
I'm using Yosys equivalence checking to help validate rewritten verilog RTL.

My issue is that converting this if-else
```
if (ai | bi) begin
    yo <= 1'b1;
end else begin
    yo <= 1'b0;
end
```
to this
```
yo <= (ai | bi);
```
results in an unproven $equiv cell.

The two versions of the code containing the above snippets are in a2.v and b2.v.  The script for Yosys processing is in eqvc2.ys and the equivalence dot file is in equiv-prep.dot.

Is this a bug or am I missing something obvious?

bobnewgard, 2017-01-30:
a2.v
```
module eqn2
    (
        input  clk_i,
        input  rst_i,
        input  ai,
        input  bi,
        output yo
    );
    reg        yo;
    always @ (posedge clk_i or posedge rst_i) begin
        if (rst_i) begin
            yo <= 1'b0;
        end else begin
            if (ai|bi) begin
                yo <= 1'b1;
            end else begin
                yo <= 1'b0;
            end
        end
    end
endmodule
```
b2.v
```
module eqn2
    (
        input  clk_i,
        input  rst_i,
        input  ai,
        input  bi,
        output yo
    );
    reg        yo;
    always @ (posedge clk_i or posedge rst_i) begin
        if (rst_i) begin
            yo <= 1'b0;
        end else begin
            yo <= (ai|bi);
        end
    end
endmodule
```
eqvc2.ys
```
read_verilog a2.v
prep -flatten -top eqn2
splitnets -ports;;
design -stash gold
read_verilog b2.v
prep -flatten -top eqn2
splitnets -ports;;
design -stash gate
design -copy-from gold -as gold eqn2
design -copy-from gate -as gate eqn2
equiv_make gold gate equiv
prep -flatten -top equiv
show -prefix equiv-prep -colors 1 -stretch
equiv_induct
equiv_status -assert
```
equiv-prep.dot
```
digraph "equiv" {
label="equiv";
rankdir="LR";
remincross=true;
n4 [ shape=octagon, label="ai", color="black", fontcolor="black" ];
n5 [ shape=diamond, label="ai_gate", color="black", fontcolor="black" ];
n6 [ shape=diamond, label="ai_gold", color="black", fontcolor="black" ];
n7 [ shape=octagon, label="bi", color="black", fontcolor="black" ];
n8 [ shape=diamond, label="bi_gate", color="black", fontcolor="black" ];
n9 [ shape=diamond, label="bi_gold", color="black", fontcolor="black" ];
n10 [ shape=octagon, label="clk_i", color="black", fontcolor="black" ];
n11 [ shape=diamond, label="clk_i_gate", color="black", fontcolor="black" ];
n12 [ shape=diamond, label="clk_i_gold", color="black", fontcolor="black" ];
n13 [ shape=octagon, label="rst_i", color="black", fontcolor="black" ];
n14 [ shape=diamond, label="rst_i_gate", color="black", fontcolor="black" ];
n15 [ shape=diamond, label="rst_i_gold", color="black", fontcolor="black" ];
n16 [ shape=octagon, label="yo", color="black", fontcolor="black" ];
n17 [ shape=diamond, label="yo_gate", color="black", fontcolor="black" ];
n18 [ shape=diamond, label="yo_gold", color="black", fontcolor="black" ];
{ rank="source"; n10; n13; n4; n7;}
{ rank="sink"; n16;}
c22 [ shape=record, label="{{<p19> A|<p20> B}|$10\n$equiv|{<p21> Y}}" ];
c23 [ shape=record, label="{{<p19> A|<p20> B}|$2_gold\n$or|{<p21> Y}}" ];
c28 [ shape=record, label="{{<p24> ARST|<p25> CLK|<p26> D}|$6_gold\n$adff|{<p27> Q}}" ];
c29 [ shape=record, label="{{<p24> ARST|<p25> CLK|<p26> D}|$9_gate\n$adff|{<p27> Q}}" ];
v0 [ label="1'0" ];
v1 [ label="1'1" ];
c31 [ shape=record, label="{{<p19> A|<p20> B|<p30> S}|$4_gold\n$mux|{<p21> Y}}" ];
x2 [shape=box, style=rounded, label="BUF"];
x3 [shape=box, style=rounded, label="BUF"];
x4 [shape=box, style=rounded, label="BUF"];
x5 [shape=box, style=rounded, label="BUF"];
x6 [shape=box, style=rounded, label="BUF"];
x7 [shape=box, style=rounded, label="BUF"];
x8 [shape=box, style=rounded, label="BUF"];
x9 [shape=box, style=rounded, label="BUF"];
n10:e -> c28:p25:w [colorscheme="dark28", color="5", fontcolor="5", label=""];
n10:e -> c29:p25:w [colorscheme="dark28", color="5", fontcolor="5", label=""];
n10:e -> x6:w:w [colorscheme="dark28", color="5", fontcolor="5", label=""];
n10:e -> x7:w:w [colorscheme="dark28", color="5", fontcolor="5", label=""];
x6:e:e -> n11:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
x7:e:e -> n12:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
n13:e -> c28:p24:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
n13:e -> c29:p24:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
n13:e -> x8:w:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
n13:e -> x9:w:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
x8:e:e -> n14:w [colorscheme="dark28", color="4", fontcolor="4", label=""];
x9:e:e -> n15:w [colorscheme="dark28", color="4", fontcolor="4", label=""];
c22:p21:e -> n16:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
c29:p27:e -> n17:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
n17:e -> c22:p20:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
c28:p27:e -> n18:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
n18:e -> c22:p19:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
n2 [ shape=point ];
c23:p21:e -> n2:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
n2:e -> c29:p26:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
n2:e -> c31:p30:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
c31:p21:e -> c28:p26:w [colorscheme="dark28", color="6", fontcolor="6", label=""];
n4:e -> c23:p19:w [colorscheme="dark28", color="6", fontcolor="6", label=""];
n4:e -> x2:w:w [colorscheme="dark28", color="6", fontcolor="6", label=""];
n4:e -> x3:w:w [colorscheme="dark28", color="6", fontcolor="6", label=""];
x2:e:e -> n5:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
x3:e:e -> n6:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
n7:e -> c23:p20:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
n7:e -> x4:w:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
n7:e -> x5:w:w [colorscheme="dark28", color="2", fontcolor="2", label=""];
x4:e:e -> n8:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
x5:e:e -> n9:w [colorscheme="dark28", color="1", fontcolor="1", label=""];
v0:e -> c31:p19:w [colorscheme="dark28", color="3", fontcolor="3", label=""];
v1:e -> c31:p20:w [colorscheme="dark28", color="8", fontcolor="8", label=""];
}
```

bobnewgard, 2017-01-30:
% yosys -V
Yosys 0.7 (git sha1 61f6811, gcc 6.2.0-10 -O2 -fdebug-prefix-map=/build/yosys-0DEyT5/yosys-0.7=. -fstack-protector-strong -fPIC -Os)
% uname -a
Linux sake 4.9.0-1-amd64 #1 SMP Debian 4.9.2-2 (2017-01-12) x86_64 GNU/Linux
%

cliffordwolf, 2017-01-30:
Here is a modified version of your script with four different methods to prove the circuit you have provided.

```
read_verilog a2.v
prep -flatten -top eqn2
splitnets -ports;;
design -stash gold
read_verilog b2.v
prep -flatten -top eqn2
splitnets -ports;;
design -stash gate
design -copy-from gold -as gold eqn2
design -copy-from gate -as gate eqn2
equiv_make gold gate equiv
prep -flatten -top equiv

opt_clean -purge
show -prefix equiv-prep -colors 1 -stretch

## method 1
opt -full
equiv_simple -seq 5
equiv_induct -seq 5
equiv_status -assert

## method 2
#equiv_struct -icells t:$adff t:$equiv
#equiv_simple -seq 5
#equiv_induct -seq 5
#equiv_status -assert

## method 3
#techmap -map +/adff2dff.v
#equiv_simple -seq 5
#equiv_induct -seq 5
#equiv_status -assert

## method 4
#clk2fflogic
#equiv_simple -seq 10
#equiv_induct -seq 10
#equiv_status -assert
```

It is generally a good idea to first run `equiv_simple` first and only then `equiv_induct`, simply because usually `equiv_simple` is faster, so it is better to prove as many `$equiv` cells as possible with that command and only then use `equiv_induct` for the remaining nodes.

The problem with your example is that the `equiv_*` commands have only limited support for asynchronous resets. Here is your equivalence checker circuit (I've cleaned it up a bit by calling `opt_clean -purge` before running `show`):

[image]

The `equiv_simple`/`equiv_induct` commands can't prove the `$adff` cells equivalent from right-to-left because to them a `$adff` cell is simply a black-box. There are a few different approaches to working around this limitation:

1. Prove the circuit from left-to-right. I.e. optimize it to a point where the two FFs are merged because the optimizer already understands that they are the same. Luckily this circuit is simple enough so that the optimizer can to this. That's what the `opt -full` does in "method 1" in the listing above.

2. Prove equivalence of the `$adff` cells structurally by assuming that if two $adff cells are driving the same outputs then they also must be driven by the same inputs. Note that this is not always true and if you try this on a circuit where this does not hold you'll just screw up your equivalence checking circuit in a way that will yield an unprovable circuit. But in this case it is ok to do that transformation. That's what the `equiv_struct -icells t:$adff t:$equiv` in "method 2" is doing.

3. Convert asynchronous resets into synchronous resets. Depending on your use case this can be a questionable approach, because it will end up proving circuits equivalent where a FF is reset asynchronously in one circuit and synchronously in the other. But maybe this is not a big deal in your case and then this can be a good approach. The call to `techmap -map +/adff2dff.v` in "method 3" does this transformation from asynchronous resets to synchronous resets.

4. Convert all clocks and asynchronous events to "fflogic", a circuit with "clockless FFs" that are driven by an implicit clock that is assumed to run fast enough to capture every relevant event on any signals. This approach potentially creates the most complex intermediate circuits to prove, and you potentially have to double the depths of all bounded methods, but it is also the most universal approach if you have to prove any properties of circuits sensitive to asynchronous events. The `clk2fflogic` command in "method 4" performs this transformation and thus creates a provable circuit.

I hope that helps. Don't hesitate to ask followup questions. Closing the issue for now.

closed 2017-01-30 by cliffordwolf

bobnewgard, 2017-01-30:
So much good stuff in your answer.  Thank you!

EmJunaid, 2022-09-14:
I'm comparing two netlists through equiv_make given as below:
**First netlist:**

module inverter(in, out);
  input in;
  wire in;
  output out;
  wire out;
  sky130_fd_sc_hd__inv_2 _0_ (
    .A(in),
    .Y(out)
  );
endmodule

**second_netlist**

module inverter (in,
    out);
 input in;
 output out;

 wire net1;
 wire net2;

 sky130_fd_sc_hd__decap_3 PHY_0 ();
 sky130_fd_sc_hd__decap_3 PHY_1 ();
 sky130_fd_sc_hd__decap_3 PHY_10 ();
 sky130_fd_sc_hd__decap_3 PHY_11 ();
 sky130_fd_sc_hd__decap_3 PHY_12 ();
 sky130_fd_sc_hd__decap_3 PHY_13 ();
 sky130_fd_sc_hd__decap_3 PHY_14 ();
 sky130_fd_sc_hd__decap_3 PHY_15 ();
 sky130_fd_sc_hd__decap_3 PHY_16 ();
 sky130_fd_sc_hd__decap_3 PHY_17 ();
 sky130_fd_sc_hd__decap_3 PHY_18 ();
 sky130_fd_sc_hd__decap_3 PHY_19 ();
 sky130_fd_sc_hd__decap_3 PHY_2 ();
 sky130_fd_sc_hd__decap_3 PHY_20 ();
 sky130_fd_sc_hd__decap_3 PHY_21 ();
 sky130_fd_sc_hd__decap_3 PHY_22 ();
 sky130_fd_sc_hd__decap_3 PHY_23 ();
 sky130_fd_sc_hd__decap_3 PHY_24 ();
 sky130_fd_sc_hd__decap_3 PHY_25 ();
 sky130_fd_sc_hd__decap_3 PHY_3 ();
 sky130_fd_sc_hd__decap_3 PHY_4 ();
 sky130_fd_sc_hd__decap_3 PHY_5 ();
 sky130_fd_sc_hd__decap_3 PHY_6 ();
 sky130_fd_sc_hd__decap_3 PHY_7 ();
 sky130_fd_sc_hd__decap_3 PHY_8 ();
 sky130_fd_sc_hd__decap_3 PHY_9 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_26 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_27 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_28 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_29 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_30 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_31 ();
 sky130_fd_sc_hd__tapvpwrvgnd_1 TAP_32 ();
 sky130_fd_sc_hd__inv_2 _0_ (.A(net1),
    .Y(net2));
 sky130_fd_sc_hd__buf_1 input1 (.A(in),
    .X(net1));
 sky130_fd_sc_hd__buf_4 output2 (.A(net2),
    .X(out));
endmodule

**Script for equivalence checking is given as:**

read_liberty -nooverwrite -lib -ignore_miss_dir -setattr blackbox cell.lib
read_verilog inverter.resized.v.without_power_pins.v
hierarchy -auto-top
hierarchy -generate sky130_fd_sc_hd__buf*
hierarchy -generate  sky130_fd_sc_hd__decap*
hierarchy -generate  sky130_fd_sc_hd__tapvpwrvgnd*
prep -flatten
splitnets -ports;;
rename -top gold
design -stash gold

read_liberty -nooverwrite -lib -ignore_miss_dir -setattr blackbox cell.lib
read_verilog inverter.v
hierarchy -auto-top
prep -flatten 
splitnets -ports;;
rename -top gate
design -stash gate

read_liberty -nooverwrite -lib -ignore_miss_dir -setattr blackbox cell.lib
design -copy-from gold -as gold gold
design -copy-from gate -as gate gate
equiv_make gold gate equiv
prep -flatten -top equiv

opt_clean -purge
show -prefix equiv-prep -colors 1 -stretch

opt -full
equiv_simple -seq 5
equiv_induct -seq 5
equiv_status -assert

**[image]**

And getting this **ERROR**:

ERROR: Found 2 unproven $equiv cells in 'equiv_status -assert'

@cliffordwolf can you tell me why I'm getting this ERROR.