YosysHQ/yosys #305: What about chisel-native support?

rgwan, 2017-02-10:
It seems chisel will be widely used in opensource hardware design. What about chisel-native support for yosys? Maybe it can improve compile efficiency :)

Martoni, 2017-02-10:
Chisel3 already have a synthesis stage indeed Chisel3 translate in FIRRTL language. Maybe the chisel support should be FIRRTL support no ?

cliffordwolf, 2017-02-11:
I don't think there is much utility to writing a chisel front-end for Yosys. The Chisel -> FIRRTL -> Verilog -> Yosys flow seems to be working fine.

There is a FIRRTL back-end under construction (`write_firrtl`), that allows you to export designs from Yosys in the FIRRTL format used by Chisel3. I'm waiting for @azidar to continue the work on it.. Adam: Is this still something you want to work on?

closed 2017-02-11 by cliffordwolf