YosysHQ/yosys #311: Top level io ports end up unconnected

klammerj, 2017-02-12:
strangely, It's only the BIBUFs. 
stuff like this:
```
          (net (rename id14732 "B2.PAD[13]") (joined
            (portRef PAD (instanceRef id14639))
          ))
```
wot I believe should look like this:
```
          (net (rename id14732 "B2.PAD[13]") (joined
            (portRef PAD (instanceRef id14639))
            (portRef (member ram_dataX 13))
          ))
```

The whole setup is here:
https://github.com/klammerj/y40k
(I've uploaded the IDS logfile, too)

klammerj, 2017-02-12:
I have now changed the verilog file to get rid of the inouts in the submodule to see if things change. 
They did not.
Then hand edited the edf to wire up the PADs. 
I am left with error messages about unconnected input ports. 
It seems to be the AN2s from the AIG pass. 
Looking at one of them the A input seems unconnected...
And a single unconnected mcu_addr[0] (this time an internal net is dangling)...

cliffordwolf, 2017-02-13:
Fixed problem described in the OP in commit db7314b.

I simply do not understand what you are talking about in your second post. Please provide a test case (minimal Yosys Script + Verilog Input) and describe in what way the expected behavior differs from the actual behavior, if the problem is still present with current Yosys git head.

closed 2017-02-13 by cliffordwolf