YosysHQ/yosys #312: aig nodes end up unconnected

klammerj, 2017-02-13:
After the latest commit I'm still getting 33 unconnected and nodes. 
They seem to be from the aigmap. 
Might be the HILOMAP.. not sure...
the logfile from the fitter is here:
https://raw.githubusercontent.com/klammerj/y40k/master/big_test.log
(it lists the primitive names)
the edf file is here:
https://raw.githubusercontent.com/klammerj/y40k/master/test/big_test.edf

cliffordwolf, 2017-02-13:
It does not look like those cells are there when I try it with current git head (69468d5):

```
yosys> stat

20. Printing statistics.

=== big_test ===

   Number of wires:                 50
   Number of wire bits:            266
   Number of public wires:          50
   Number of public wire bits:     266
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                101
     BIBUF                          32
     GCLKBUF                         2
     IBUF                           28
     OBUF                           37
     RSBUF                           1
     djsdram_simple1                 1

=== djsdram_simple1 ===

   Number of wires:               6675
   Number of wire bits:           7436
   Number of public wires:          50
   Number of public wire bits:     291
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               7275
     AN2                          2401
     FD                             86
     FDE                            33
     INV                          3826
     ONE                           345
     ZERO                          584

yosys> check -noinit
21. Executing CHECK pass (checking for obvious problems).
checking module big_test..
checking module djsdram_simple1..
found and reported 0 problems.

yosys> select -assert-none i:* t:ZERO t:ONE %% %co* t:AN2 %D

yosys> write_edif -nogndvcc ./test/big_test.edf

22. Executing EDIF backend.

yosys> exit
```

(The `select -assert-none i:* t:ZERO t:ONE %% %co* t:AN2 %D` should fail if there are any unconnected `AN2` cells.)

Please check if updating to git head fixes the issue for you.

klammerj, 2017-02-14:
Yes, I'm at head too.
I don't know if they're connected to anything in the internal rep. 
('twas merely a guess wrt the hilomap... could be some arithmetic primitive too as I 've got no cells for those yet.. or te aigmap)
I only know it's unconnected in the edf file...
Grepping for 15552 gets me to id00995(@line 2148), 
and grepping for that finds a connected B(@line 19807) and Q(@line 19832) but no A. 
I guess it could also be sthg in the .v file.. will read a bit...

klammerj, 2017-02-14:
PS: I've made changes to the .v during the previous bug report, when I tried to get rid of a warning about internal tristates. I just now changed some port types from reg to wire. so maybe you want to pull the testcase again. I've also uploaded the logfile from IDS (big_test.log). and the output I get ./test/big_test.edf.

cliffordwolf, 2017-02-14:
For some reason the EDIF file I get is slightly different than yours. Can you also upload the log ouput produced by yosys so I can figure out what's different? (Yosys avoid non-deterministic programming pattern and should produce the same output for the same input on all platforms and compilers.)

I think your `id00995` is my `id00994`:

          (instance (rename id00994 "$auto$aigmap.cc:105:execute$15548")
            (viewRef VIEW_NETLIST (cellRef AN2 (libraryRef LIB))))
         ....
         (net (rename id08550 "$auto$rtlil.cc:1733:NotGate$15547") (joined
            (portRef B (instanceRef id00994))
            (portRef QN (instanceRef id03827))
          ))
          (net (rename id08551 "$auto$rtlil.cc:1734:AndGate$15549") (joined
            (portRef A (instanceRef id03828))
            (portRef Q (instanceRef id00994))
          ))

No connection on port A. When I let Yosys create a `.il` file as well (`write_ilang output.il`) I can see why:

```
  attribute \src "cells_map.v:100"
  cell \AN2 $auto$aigmap.cc:105:execute$15548
    connect \A 1'x
    connect \B $auto$rtlil.cc:1733:NotGate$15547
    connect \Q $auto$rtlil.cc:1734:AndGate$15549
  end
```

The A input is driven by an undef. This AN2 is in the cone driving `refresh_running`:

    yosys [djsdram_simple1]> show $auto$aigmap.cc:105:execute$15548 %co20:-FDE %co2

[image]

This undef is added by the `dff2dffe` pass. Here is before `dff2dffe`:

[image]

And this is after:

[image]

This transformation is correct afaict. Running for example `opt_expr -mux_undef` will get rid of the MUX:

[image]

Note that for example `synth_ice40` does run `opt_expr -mux_undef` shortly after calling `dff2dffe`:

```
    map_ffs:
        dffsr2dff
        dff2dffe -direct-match $_DFF_*
        techmap -map +/ice40/cells_map.v
        opt_expr -mux_undef
        simplemap
        ice40_ffinit
        ice40_ffssr
        ice40_opt -full
```

I'm now closing this issue. Add a comment below if and when you upload your yosys log output. I'll use it to check if I can explain why you get a slightly different EDIF file than I am.

closed 2017-02-14 by cliffordwolf

cliffordwolf, 2017-02-14:
fyi: commit 4e80ce9 adds warnings for cell ports driven by x/z to `write_edif`.

klammerj, 2017-02-14:
I've uploaded the log now

klammerj, 2017-02-14:
PS: Don't worry about it too much. The box here is 32 bit, and that may well be the cause of it. I remember one of your bigsim tests failing because it overflowed an int here.