YosysHQ/yosys #315: EDIF port endianness trouble

klammerj, 2017-02-15:
I just noticed after importing that the mcu_addrX bits got wired in reverse. 
changing the port decl to:
```
          (port (array (rename mcu_addrX "mcu_addrX[0:23]") 24) (direction INPUT))
```
fixes the thing, although the mcu_addrX is declared as 
```
input  [23:0] mcu_addrX,
```
in the verilog file.
unfortunately I can't tell which is in error the write_edif or 
IDS. when researching this the only vaguely similar thing I came across was:
https://www.xilinx.com/support/answers/5416.html
unfortunately there's no hint there which bit a
```
            (portRef (member mcu_addrX 1))
```
is supposed to connect to...
(there's at least two interpretations for this that I can think of)

klammerj, 2017-02-15:
Here's a patch what fixes it for me:
[0001-put-rename-clauses-in-module-ports-for-less-ambiguit.patch.txt](https://github.com/cliffordwolf/yosys/files/778104/0001-put-rename-clauses-in-module-ports-for-less-ambiguit.patch.txt)

klammerj, 2017-02-15:
Here's another one making this stuff configurable via parameters. 
It'll still fight with the name munging code... I don't have the big plan how to integrate it properly. 

[0002-add-options-for~f-flavors.patch.txt](https://github.com/cliffordwolf/yosys/files/778537/0002-add-options-for.f-flavors.patch.txt)

cliffordwolf, 2017-02-23:
> 0001-put-rename-clauses-in-module-ports-for-less-ambiguit.patch.txt
> 0002-add-options-for~f-flavors.patch.txt

Can you create a proper PR for this so we can use the usual github methods for reviewing a patch?

EDIF_DEF() takes a name and produces a `(rename ...)` clause if necessary. You take its result and treat it like its always a simple identifier.

I.e. consider the following code:

```
module test(input [3:0] \crazy%names , output [0:3] \break%your%patch );
  assign o = i;
endmodule
```

And run it through `yosys -p 'prep; write_edif -pidx up' test.v`. The result contains this two lines:

          (port (array (rename (rename id00001 "break%your%patch") "(rename id00001 "break%your%patch")[0:3]") 4) (direction OUTPUT))
          (port (array (rename (rename id00002 "crazy%names") "(rename id00002 "crazy%names")[0:3]") 4) (direction INPUT))

Also: RTLIL Wires have `upto` and `start_offset` properties that preserve all neccessary information about the original ranges. Use them! As long as you are assuming that everything is homogeneously upto or downto, and/or simply assume that everything starts/ends at index zero, your code isn't really fixing anything.

Also: For hierarchical designs you must make sure that you do not break the interfaces at the module boundaries.

Please address those issues and create a proper PR.

closed 2017-02-23 by cliffordwolf