YosysHQ/yosys #325: Constant zeros as input in $logic_not

Kmanfi, 2017-02-25:
Concatenated constant zeros in input A can be removed.
In this special case, after removing constant, input width be then 1, when the cell could be changed to $not, which might enable further optimisation.

```
       "_10298_": {
          "hide_name": 1,
          "type": "$logic_not",
          "parameters": {
            "A_SIGNED": 0,
            "A_WIDTH": 2,
            "Y_WIDTH": 1
          },
          "port_directions": {
            "A": "input",
            "Y": "output"
          },
          "connections": {
            "A" : [["_02009_", 0], [0]],
            "Y" : ["_03931_"]
          }
        },
```

cliffordwolf, 2017-02-25:
What am I looking at? This `"connections": {...}` block is not the syntax Yosys' `write_json` produces.

Kmanfi, 2017-02-25:
Ah, oops, that's true. Its output from our own back end. Very similar as write_json.
I will write same design with write_json and get back to this.

Kmanfi, 2017-02-25:
Not exactly the same gate as above but same idea.
```
        "$techmap\\fooo”: {
          "hide_name": 1,
          "type": "$logic_not",
          "parameters": {
            "A_SIGNED": 0,
            "A_WIDTH": 2,
            "Y_WIDTH": 1
          },
          "attributes": {
            "src": “food”
          },
          "port_directions": {
            "A": "input",
            "Y": "output"
          },
          "connections": {
            "A": [ 27754, "0" ],
            "Y": [ 27755 ]
          }
        },

```

Kmanfi, 2017-02-25:
Could be opted to single $not.
```
        "$techmap\\foooo.$procmux$70841_CMP0": {
          "hide_name": 1,
          "type": "$logic_not",
          "parameters": {
            "A_SIGNED": 0,
            "A_WIDTH": 2,
            "Y_WIDTH": 1
          },
          "attributes": {
            "src": “fooo.v:274510"
          },
          "port_directions": {
            "A": "input",
            "Y": "output"
          },
          "connections": {
            "A": [ 1474, 1474 ],
            "Y": [ 15330 ]
          }
        },

```

Kmanfi, 2017-07-26:
Close this due to "logical mistake".

closed 2017-07-26 by Kmanfi