YosysHQ/yosys #330: ICE40 DDR output simulation not matching oscope

dlharmon, 2017-03-10:
The output of Icarus/GTKWave using cells_sim.v matches what I'd expect from the ICE tech library document. 

When synthesized and loaded in hardware, it appears as if D_OUT_0 and D_OUT_1 were swapped.

Simplified code to demonstrate issue: 
```
reg [1:0] ledi;
   always @ (posedge clockbuf)
     begin
        case(count[5:0])
          0: ledi <= 2'b00;
          1: ledi <= 2'b11;
          2: ledi <= 2'b01;
          3: ledi <= 2'b10;
          4: ledi <= 2'b00;
          5: ledi <= 2'b11;
          6: ledi <= 2'b00;

          default: ledi <= 2'b11;
        endcase
     end

   SB_IO #(.PIN_TYPE(6'b110000), .IO_STANDARD("SB_LVCMOS")) led_ddr
     (.PACKAGE_PIN(led),
      .CLOCK_ENABLE(1'b1),
      .OUTPUT_CLK(clockbuf),
      .OUTPUT_ENABLE(1'b1),
      .D_OUT_0(ledi[0]), // data out to pin                                                                                              
      .D_OUT_1(ledi[1]));
```

[image]

[image]

Note the difference in the order of the output bits.
00 11 01 10 00 11 00 for simulation
00 11 _10 01_ 00 11 00 in hardware

led and clockbuf in the GTKWave trace are the two signals on the scope.

The chip is an ICE40HX1K.
I'm currently planning to work around it by doubling the clock frequency.

cliffordwolf, 2017-03-11:
Is it possible that this is related to https://github.com/cseed/arachne-pnr/issues/61 and is fixed by https://github.com/cseed/arachne-pnr/pull/62?

Btw: Less than an [mcve](http://stackoverflow.com/help/mcve) is not very helpful. Just posting code snippets instead of a minimal example that I can actually run prevents me from reproducing the thing you are observing.

dlharmon, 2017-03-11:
It was fixed by https://github.com/cseed/arachne-pnr/pull/62. Thanks!

closed 2017-03-11 by dlharmon